mavlink-router/mavlink-router #371: Issue with several dynamic components in one system

jbeyerstedt, 2022-08-04:
Hello, I'm kinda late to the party but I am feeling the pain of not having support for the following scenario, related to what is being discussed here..:

I have a system that manages several cameras as MAVLink components, which are dynamically created/deleted in runtime:
```
                                   .--------.
                                   | sys_1  |
                                   | --/\-- |
[QGC] <---> [mavlink-router] <---> | comp_1 |
                                   | comp_2 |
                                   | comp_3 |
                                   '--------'
```

In this case, `QGC` receives the heartbeats from every component of `sys_1`, but when it tries to talk with `comp_1`, it eventually succeeds but usually fails because `mavlink-router` is routing all messages to the last component it has received a heartbeat from, so sometimes `comp_3` is receiving messages sent from QGC for `comp_1` and `comp_2`,  sometimes `comp_1` is receiving it all, and the same happens with `comp_2`.

_Originally posted by @joaoantoniocardoso in https://github.com/mavlink-router/mavlink-router/issues/350#issuecomment-1205280924_

jbeyerstedt, 2022-08-04:
@joaoantoniocardoso: To understand your issue: Do you have one software talking to mavlink-router or one application per camera? If it's one per camera, what kind of endpoint are you using in mavlink-router?

Because this should just work when using the TCP server port of mavlink-router. That way each camera would have it's own endpoint in mavlink-routers routing logic and each component should only get data that's intended for it's component ID.

Having multiple systems "connect" to one UDP "server" mode endpoint will not work though as all cameras will share one endpoint. And an UDP endpoint will always send outgoing data to the IP and port from which the last incoming message on that endpoint was received.

joaoantoniocardoso, 2022-08-04:
Perfect, thanks for your insight @jbeyerstedt, I was in fact using UDP, by changing it to use TCP (at least in the current `master`) works!

We can close it.

closed 2022-08-04 by lucasdemarchi