mavlink-router/mavlink-router #373: crash program in uart endpoint

mohammada315, 2022-08-09:
dear mavlink-router developers
I need to send a uart stream to some other telemetry
for example
                               
```
#[UartEndpoint echo]
#Device = /dev/serial/by-id/usb-Holybro_Durandal_2C0024001151303138363434-if00
#Baud = 115200

#[UartEndpoint alpha]
#Device = /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_D306NG9H-if00-port0 
#Baud = 115200
```
But after a few moments, the program exits

mark-nick-o, 2022-08-31:
run the mavlink-routerd from the command line after stopping it with a ps -ael | grep mavlink-routerd and look at the output
if you have no usb device attached you would get something like this

[mav-sony@mav-sony ~]$ mavlink-routerd
mavlink-router version v2-235-gcf93476
Could not open /dev/ttyACM0 (No such file or directory)

lucasdemarchi, 2022-09-01:
The uart link is considered a "critical" link. If it disappears, mavlink-router considers it can't function normally anymore and exits with an error. The systemd service is supposed to restart it when it exits with an error, so if it can reset and open the uart again, it would recover. However if the device is really not available anymore, there isn't much that can be done. 

The device disappearing is usually one of:

1) being disconnected, in the case of USB
2) kernel bug - update your kernel to the latest stable release
3) hardware bugs - there may be workarounds in newer kernel versions

jbeyerstedt, 2024-12-10:
Closing because of no activity and the observed behavior was the intended behavior. Feel free to comment and re-open, if there are things left to discuss.

closed 2024-12-10 by jbeyerstedt