mavlink-router/mavlink-router #374: mavlink application program closes when mavlink-routerd stops

mark-nick-o, 2022-08-29:
hello
a totally different piece of code that i used before last time we spoke
last time it was also 32bit rasbian OS and we did kernel upgrade to fix it

this time 64bit manjaro OS completely different

same physical raspberry pi 4 though.

this is the piece of code in question its operated from a timed interrupt
the command sends a request message on mavlink via mavlink-routerd to an FCU

if i compile with this function in the program, it crashes with no explanation if mavlink-routerd stops running....... and also sometimes if i connect/re-conect the mission planner with mavlink-routerd still running and not stopping.

void alrm( int signo )
{
    // set slow rate on servo output
    int st = clock();
    SendComRequestSERVO_OUTPUT_RAWInterval(system_ID, 0, 1000000000000000.0f);
    int en = clock();
    std::cout << "action time from alarm handler : " << (en-st)/double(CLOCKS_PER_SEC)*1000.0f << "\n";
    //std::cout << "handler interrupting only !! : " << (en-st)/double(CLOCKS_PER_SEC)*1000.0f << "\n";
}

if i change it to this it doesn't die ever (no message but still connected to end point)
void alrm( int signo )
{
    // set slow rate on servo output
    int st = clock();
    //SendComRequestSERVO_OUTPUT_RAWInterval(system_ID, 0, 1000000000000000.0f);
    int en = clock();
    //std::cout << "action time from alarm handler : " << (en-st)/double(CLOCKS_PER_SEC)*1000.0f << "\n";
    std::cout << "handler interrupting only !! : " << (en-st)/double(CLOCKS_PER_SEC)*1000.0f << "\n";
}

the mavlink routine it calls is like this....................
static void SendComRequestSERVO_OUTPUT_RAWInterval (uint8_t target_sys, uint8_t target_comp, float rate);
static void SendComRequestSERVO_OUTPUT_RAWInterval (uint8_t target_sys, uint8_t target_comp, float rate)
{
 /*
send command long over mavlink to change rate of servo message
 */
  mavlink_command_long_t command = { 0u };                                          // Command Type
  mavlink_message_t msg;                                                        // encoder type
  command.target_system    = target_sys;
  command.target_component = target_comp;
  command.command          = MAV_CMD_SET_MESSAGE_INTERVAL;
  //command.confirmation     = target_comp;
  command.param1           = (float)(MAVLINK_MSG_ID_SERVO_OUTPUT_RAW);                       
  command.param2           = rate;                                                // 10 Hz
  /* encode */
  mavlink_msg_command_long_encode(target_sys, target_comp, &msg, &command);

  /* Copy the message to send buffer when buf is uint8_t buf[MAVLINK_MSG_ID_COMMAND_LONG_LEN]; */
  msg_send(&msg);
  std::cout<<"> Requesting Servo Data\n";
}

any idea why ?

i have the following exception handler it never sees one when this program dies.

static void setup_signal_handlers()
{
    struct sigaction sa = { };
    std::cout<<"> Setting up signal handlers"<<std::endl;
    sa.sa_flags = SA_NOCLDSTOP;
    sa.sa_handler = exit_signal_handler;
    #if defined(_MSC_VER)
         //Allow floating-point exceptions
        _control87( 0, _MCW_EM );
    #endif
    sigaction(SIGTERM, &sa, NULL);
    sigaction(SIGINT, &sa, NULL); 
    sigaction(SIGQUIT,&sa, NULL);
    sigaction(SIGFPE,&sa,NULL);
    sigaction(SIGHUP,&sa,NULL);
    /* we are using alarm for timer so surpress it here sigaction(SIGALRM,&sa,NULL ); */
    sigaction(SIGSEGV,&sa,NULL); 
    sigaction(SIGTSTP,&sa,NULL);      
}

the exit for that signal handler never prints anything......

also there is error handling on the sockets and they dont indicate any errors either

mark-nick-o, 2022-08-30:
i've ran this sucessfully after compiling cpulimit for the manjaro core and limiting this and the application to 70% but im interested to know if you have any comments on it. I think the crashes are down to cpu of Raspberry Pi4 and mavlink-routerd was getting over 100% and when they are both limited neither went down and operated with 3 GCS connected (2x Mission Planner 1x QGC).

mark-nick-o, 2022-08-30:
maybe spoke too soon
then mavlink-routerd went down like this

Endpoint [16] accepted message 245 to -1/-1 from 2/1
TCP [16]dynamic: Wrote 13 bytes
Endpoint [4]to_fc: got message 148 to -1/-1 from 2/1
        Known components:
                2/1
                1/0
                253/36
                28/0
                219/2
                58/14
                184/2
                2/253
Endpoint [5]to_CAM_100: got message 148 to -1/-1 from 2/1
        Known components:
Endpoint [5] accepted message 148 to -1/-1 from 2/1
UDP [5]to_CAM_100: Wrote 71 bytes
Endpoint [7]to_mavproxy: got message 148 to -1/-1 from 2/1
        Known components:
Endpoint [7] accepted message 148 to -1/-1 from 2/1
UDP [7]to_mavproxy: Wrote 71 bytes
Endpoint [-1]to_fc: g

mark-nick-o, 2022-08-30:
on the application end i got the message as expected now

action time from alarm handler : 0.129
> Requesting Servo Data
action time from alarm handler : 0.143
Error reading socket (Connection reset by peer)
Process 73424 terminated abnormally

mark-nick-o, 2022-08-30:
trying again with
sudo cpulimit --limit=60 mavlink-routerd

mark-nick-o, 2022-08-30:
no sadly the connection from the GCS was slow and it went down again very similiar

tail of mavlink-routerd
TCP [15]dynamic: Wrote 36 bytes
Endpoint [16]dynamic: got message 26 to -1/-1 from 2/1
        Known components:
                215/190
Endpoint [16] accepted message 26 to -1/-1 from 2/1
TCP [16]dynamic: Wrote 36 bytes
Endpoint [17]dynamic: got message 26 to -1/-1 from 2/1
        Known components:
                255/190
Endpoint [17] accepted message 26 to -1/-1 from 2/1
TCP [17]dynamic: Wrote 36 bytes
poll error for fd 4
Critical fd 4 got error, exiting
TCP [13]dynamic: Connection closed
TCP [14]dynamic: Connection closed
TCP [15]dynamic: Connection closed
TCP [16]dynamic: Connection closed
TCP [17]dynamic: Connection closed

can these prints be disabled perhaps they are creating a load ?

app said this time
> Requesting Servo Data
action time from alarm handler : 0.098
> Requesting Servo Data
action time from alarm handler : 0.16
Process 76922 terminated abnormally

no socket error

top shows both at 60% cpu

mark-nick-o, 2022-08-30:
i tried another cpu entirely x86 unbuntu 18
Linux Ubunt-18-dev1 5.4.0-122-generic #138~18.04.1-Ubuntu SMP Fri Jun 24 14:14:03 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

and another different piece of code not being activated by timed interrupts

cpu was 100% again

same happened
app 👍 
*******************MAV_CMD_REQUEST_MESSAGE**************Command Long cmd=521 p1=0.000000send Heart-beat : 999.781
heartbeat took : 0.059
send Heart-beat : 999.797
heartbeat took : 0.058
********************MAV_CMD_REQUEST_MESSAGE**************Command Long cmd=521 p1=0.000000send Heart-beat : 999.795
heartbeat took : 0.056
send Heart-beat : 999.784
heartbeat took : 0.062
send Heart-beat : 999.776
heartbeat took : 0.055
send Heart-beat : 999.574
heartbeat took : 0.051
send Heart-beat : 999.742
heartbeat took : 0.04
send Heart-beat : 999.707
heartbeat took : 0.037
send Heart-beat : 999.806
heartbeat took : 0.037
send Heart-beat : 999.86
heartbeat took : 0.061
send Heart-beat : 999.876
anthony@Ubunt-18-dev1:~/ABD-MAV/build$

mavlink-routerd
Endpoint [17] got message 4294967295 to 2/0 from 2/0
        Known components:
                215/190
Endpoint [18] got message 4294967295 to 2/0 from 2/0
        Known components:
                215/190
Endpoint [19] got message 4294967295 to 2/0 from 2/0
        Known components:
                255/190
Endpoint [16] got message 4294967295 to 2/0 from 2/0
        Known components:
                2/100
Endpoint [16] accepted message 4294967295 to 2/0 from 2/0
TCP [16] wrote 19 bytes
TCP Connection [17] closed
TCP Connection [18] closed
TCP Connection [19] closed
TCP Connection [16] closed

mark-nick-o, 2022-08-30:
saw this one

Endpoint [16] accepted message 251 to -1/-1 from 2/1
TCP [16] wrote 23 bytes
poll error for fd 18, closing it
Endpoint [5] got message 4294967295 to 2/0 from 2/0
        Known components:
Endpoint [6] got message 4294967295 to 2/0 from 2/0
        Known components:
Endpoint [7] got message 4294967295 to 2/0 from 2/0
        Known components:
Endpoint [8] got message 4294967295 to 2/0 from 2/0
        Known components:
Endpoint [9] got message 4294967295 to 2/0 from 2/0
        Known components:
                2/1
                1/0
Endpoint [9] accepted message 4294967295 to 2/0 from 2/0
UART [9] wrote 19 bytes
Endpoint [-1] got message 4294967295 to 2/0 from 2/0
        Known components:
                2/0
Endpoint [19] got message 4294967295 to 2/0 from 2/0
        Known components:
                255/190
Endpoint [18] got message 4294967295 to 2/0 from 2/0
        Known components:
                215/190
Endpoint [17] got message 4294967295 to 2/0 from 2/0
        Known components:
                215/190
Endpoint [16] got message 4294967295 to 2/0 from 2/0
        Known components:
                2/100
Endpoint [16] accepted message 4294967295 to 2/0 from 2/0
TCP [16] wrote 19 bytes
TCP Connection [19] closed
TCP Connection [18] closed
TCP Connection [17] closed
TCP Connection [16] closed
anthony@Ubunt-18-dev1:~$

mark-nick-o, 2022-08-30:
on tracing this error i have found this difference

mainloop.cpp (on the x86 box built a while back i think)

            if (events[i].events & EPOLLERR) {
                log_error("poll error for fd %i, closing it", p->fd);
                remove_fd(p->fd);
                // make poll errors fatal so that an external component can
                // restart mavlink-router
                request_exit();
            }
			
mainloop.cpp (on github)

            if (events[i].events & EPOLLERR) {
                log_error("poll error for fd %i", p->fd);

                if (p->is_critical()) {
                    log_error("Critical fd %i got error, exiting", p->fd);
                    request_exit(EXIT_FAILURE);
                } else {
                    log_debug("Non-critical fd %i, error is okay.", p->fd);
                }
            }
			
on raspberry pi built (2 days ago from a git clone)
there is no source code to check
common  mavlink-routerd  mavlink-routerd.p

./common:
libcommon-private.a  libcommon-private.a.p

./common/libcommon-private.a.p:
conf_file.cpp.o  log.cpp.o  util.cpp.o  xtermios.cpp.o

./mavlink-routerd.p:
autolog.cpp.o  binlog.cpp.o  dedup.cpp.o  endpoint.cpp.o  git_version.cpp.o  logendpoint.cpp.o  main.cpp.o  mainloop.cpp.o  pollable.cpp.o  timeout.cpp.o  ulog.cpp.o

mark-nick-o, 2022-08-30:
changing the x86 mavlink-routerd to one compiled from a .zip source code here then gives me 4% cpu load for mavlink-routerd but doesnt seem to perform the commands in the application connected to it (which is at cpu 100%)

mark-nick-o, 2022-08-31:
after the following kernel tweaks it seems to run very well (didnt stop in a whole day of testing although i said this before)

this is using the version of source code from the .zip and building on intel x86 board

hope this info is useful

========================= disabling powersave mode in x86 ====================================================

cat /sys/devices/system/cpu/cpuidle/current_driver
intel_idle

vi /etc/default/grub
GRUB_CMDLINE_LINUX="intel_idle.max_cstate=0 processor.max_cstate=0"

sudo update-grub

systemctl reboot

then you should get

you@Ubunt-18-dev1:~$ cat /sys/devices/system/cpu/cpuidle/current_driver
none

========================= Notes for x86 Performance Improvement ====================================================

you@Ubunt-18-dev1:~$ sudo cpupower info
analyzing CPU 0:
perf-bias: 6

you@Ubunt-18 $ sudo cpupower frequency-info

>>>>>>>>>>>>>>>>>>> BEFORE <<<<<<<<<<<<<<<<<<
you@Ubunt-18-dev1:~$ sudo cpupower frequency-info
analyzing CPU 0:
  driver: intel_pstate
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 500 MHz - 1.90 GHz
  available cpufreq governors: performance powersave
  current policy: frequency should be within 500 MHz and 1.90 GHz.
                  The governor "powersave" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 1.23 GHz (asserted by call to kernel)
  boost state support:
    Supported: no
    Active: no
	
>>>>>>>>>>>>>>>>>> AFTER <<<<<<<<<<<<<<<<<<<<<<<<<
you@Ubunt-18-dev1:~$ sudo cpupower frequency-info
analyzing CPU 0:
  driver: intel_pstate
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 500 MHz - 1.90 GHz
  available cpufreq governors: performance powersave
  current policy: frequency should be within 500 MHz and 2.00 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 1.51 GHz (asserted by call to kernel)
  boost state support:
    Supported: yes
    Active: yes

------------------------------------- cpu turbo boost enabling ------------------------

sudo apt install msr-tools
[... 109 lines cut ...]
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:
 
============================= manjaro linux RPI ==================================

[mav-sony@mav-sony ~]$ cat /sys/devices/system/cpu/cpuidle/current_driver
none

available govenors
[mav-sony@mav-sony ~]$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
conservative ondemand userspace powersave performance schedutil

[mav-sony@mav-sony ~]$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies
600000 700000 800000 900000 1000000 1100000 1200000 1300000 1400000 1500000

echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

add this to /boot/config.txt to overclock it# Overclock CPU
over_voltage=4
arm_freq=1900

after this you should have
[mav-sony@mav-sony ~]$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies
600000 700000 800000 900000 1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000

mark-nick-o, 2022-09-01:
runs perfect on x86 4x Mission Planner and 1x QGC connected for 48hrs and very fast to get commands as you say so long as no link goes to sleep and the kernel is optimised it runs good

jbeyerstedt, 2023-02-18:
Could you please summarize what the issue is? Your big block of information is hard to follow, especially because it's not formatted properly.
Is there a bug in mavlink-router to be fixed or does it work fine for you now?

jbeyerstedt, 2023-12-07:
Closing because of no response.

closed 2023-12-07 by jbeyerstedt