mavlink-router/mavlink-router #376: 4G，Connect to MP or QGC, getting parameters stuck at the end

zouzhe1, 2022-12-06:
# problem
Connect to MP or QGC, getting parameters stuck at the end[STAT_RYNTIME]
[image]
---
# My connection diagram is as follows
[image]

---

1，My 4GDTU module uses UDP6008 port to forward the pixhawk serial port data to the cloud server
2，The cloud server forwards UDP 6008 to UDP 6009

`./mavlink-routerd -e 127.0.0.1:6009 0.0.0.0:6008`
3，**PC mission planner UDP connection mode, only the port number can be selected, and the IP address cannot be filled in，WHY？？**
4，So I choose TCP 5760 communication in [ 4G DTU ]--[ clould server]--[ PC ]
5，I also tried using QGC, and the connection could not get the parameters smoothly. Every time it was loaded to the end, it was stuck in **STAT_RUNTINME**

please help me,thanks

jbeyerstedt, 2022-12-09:
I only know QGC and not ArduPilot Mission Planner. But in QGC you basically have two options to get data in: Sending UDP packets to port 14550 on your PC (QGC will listen for incoming data) or by manually configuring a communication link in the settings (QGC will connect to a "server" or device via TCP/ UDP/ UART).
So that's probably the reason why can only configure a UDP port and no address as Mission Planner is the "server" and will wait for incoming data on that port.

From my experience, I would always recommend connecting to the TCP server of mavlink-router, if mavlink-router is running for longer than a client is connected. If you start mavlink-router with your rover, using an UDP endpoint will be fine, though.
(The reason is, that a TCP connection will be cleaned up properly when you disconnect, but that's not possible with an UDP endpoint.)

In theory, you setup should work though. I have a setup where I have several "server" mode UDP endpoints receiving data from drones and QGC is connected via the TCP port of mavlink-router.
As some parameters are loading, the parameter request was properly forwarded from QGC to the rover and you also have a connection in the other direction. So I don't really have an idea why this doesn't work all the way.

You could try to start mavlink-router with "-v" and try to see which MAVlink packages are transmitted between which endpoints. Or start with "-r" to get stats and see, if there are messages to "unknown endpoints". This would hint to some issue in the routing logic.
As each endpoint remembers which system is connected, it's important to have no message loops and only use each MAVLink system ID on one device or software. This is also important if different PCs are running QGC.

jbeyerstedt, 2024-12-10:
Closing because of no activity. Feel free to comment and re-open, if there are things left to discuss.

closed 2024-12-10 by jbeyerstedt