mavlink-router/mavlink-router #377: Very low read rate from LW20 lidar

matanj83, 2022-12-06:
Hi,
Can you please help or give insights with the following?

I get very low read rate (0.2-0.8 Hz) from a LW20C lidar when running mavlink-router.
I use QGroundControl's Mavlink inspector for inspecting the following:
[image]

note the low rate of several mavlink messages, along with the DISTANCE_SENSOR that I read the lidar values from.

The configuration is a Jetson NX computer that runs mavlink router, connected via UART to a PX4 (the lidar is connected via I2C).
The QGroundControl app runs on my laptop, connected by USB to the Jetson.

If I don't run mavlink router and connect directly my laptop to the PX4, I don't see those low rates:
[image]

Note that here only the battery message is in the low rate as before.

I run with the following configuration:

[General]
#Mavlink-router serves on this TCP port
TcpServerPort=0
ReportStats=false
MavlinkDialect=auto

[UartEndpoint FMU]
Device = /dev/ttyTHS0
Baud = 115200

[UdpEndpoint wonderapp]
Mode = Normal
Address = 127.0.0.1
Port = 14553
ReportStats = true

[UdpEndpoint GCS]
Mode = Normal
# Update Me
Address = 192.168.55.100
# Update Me
Port = 14550
ReportStats = true

Thanks,
Matan

matanj83, 2022-12-08:
I also see that if I remove the two UDP endpoints in my conf file (effectively having only the UART endpoint) , the DISTANCE_SENSOR message (ID 132) rate improves.

I use "DebugLogLevel = debug" in the general section of the conf file, and send stdout & stderr to a file.
I use the line numbers as primitive way to estimate the rates...
Is there a way to log also timestamps when setting debug mode?

Thanks again!

jbeyerstedt, 2022-12-09:
In PX4, there are some settings for sending telemetry data via a serial port. 115200 baud is a quite reasonable value for high UART data rate while still being reliable.

But then you assign a MAVLink instance to the telemetry port, which has some additional settings:
- MAV_x_CONFIG: Selects the telemetry port
- MAV_x_RATE: Limits the data bandwidth of the transmitted messages to 1200 Byte/s by default. Set it to 0 to automatically choose a reasonable value based on the configured UART/ Telemetry port baud rate.
- MAV_x_MODE: Defines which messages should be transmitted at which frequency.

So my guess is, that either the MAV_x_RATE parameter is set to the default value or the selected MODE is limiting your update rate. "Normal" mode is usually fine while "Onboard" has a very high frequency of sensor data for example. Unfortunately the exact config which uORB topics/ MAVLink messages are sent can only be found in the source code (at least the last time I have looked for it).

matanj83, 2022-12-11:
@jbeyerstedt , you are a life saver! Works like a charm!
Thanks for your rapid help and wonderful tool!

closed 2022-12-11 by matanj83