mavlink-router/mavlink-router #382: Connection lost after 10s consistently in any mode

kaangoksal, 2022-12-18:
Here is my conf file, nothing complex: 

```bash
pi@ras-pilot:~/mav-telemetry-linux/bin $ cat /etc/mavlink-router/main.conf 

[UartEndpoint usbUART]
Device = /dev/ttyACM0  
Baud = 115200

#[UartEndpoint pinUART]
#Device = /dev/ttyS0
#Baud = 57600

[UdpEndpoint MainUdp]
mode = Server
Address = 0.0.0.0
Port = 14555
```

```bash
pi@ras-pilot:~/mavlink-router/build/src $ ./mavlink-routerd
mavlink-router version v3
Opened UART [4]usbUART: /dev/ttyACM0
UART [4]usbUART: speed = 115200
Opened UDP Server [5]MainUdp: 0.0.0.0:14555
Opened TCP Server [6] [::]:5760
TCP [8]dynamic: Connection accepted
113 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
111 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
111 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
```

I compiled the project on a raspberry pi 4 and also raspberry pi 3. Autopilot is connected with USB, I use apm planner to connect to mavlink router, both devices are on local network. 

I see following messages on apm planner side: 
``` bash

HBD: Heartbeat stopped
Requested retransmission of #595
```

I also started watching the packets using the following command: 

`tcpdump -i wlo1 -n udp port 14555` 

I see that mavlink-router stops sending messages to apm planner shortly after apm planner starts sending messages. 

I don't have this problem with mavproxy.py, it seems like mavlink-router is stopping transmission very soon after a connection is established. I have to re-start mavlink-router to just gain 10 seconds of connection.

how should I debug this?

kaangoksal, 2022-12-18:
I ran the mavlink router in -v mode, and I see this message

```
Message 147 to unknown sysid/compid: -1/-1
> UART [4]usbUART: Got 16 bytes
Endpoint [4]usbUART: got message 11020 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [5]MainUdp: got message 11020 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 11020 to unknown sysid/compid: -1/-1
> UDP [5]MainUdp: Got 21 bytes
Endpoint [4]usbUART: got message 0 to -1/-1 from 252/1
	Known components:
		1/1
Endpoint [4] accepted message 0 to -1/-1 from 252/1
UART [4]usbUART: Wrote 21 bytes
Endpoint [5]MainUdp: got message 0 to -1/-1 from 252/1
	Known components:
		252/1
		1/1

```

so apm planner seems to be sending the message to -1/-1 instead of 1/1 ?? maybe that is the issue?

kaangoksal, 2022-12-18:
Here is the fix 
```
[UartEndpoint usbUART]
Device = /dev/ttyACM0  
Baud = 115200
MavlinkDialect = auto
AllowSrcSysIn = 1

[UdpEndpoint MainUdp]
mode = Normal
Address = 10.0.0.167
Port = 14550
AllowSrcSysIn = 252

```

result of the output 

```

Endpoint [5] accepted message 241 to -1/-1 from 1/1
UDP [5]MainUdp: Wrote 32 bytes
Endpoint [4]usbUART: got message 147 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [5]MainUdp: got message 147 to -1/-1 from 1/1
	Known components:
		252/1
Endpoint [5] accepted message 147 to -1/-1 from 1/1
UDP [5]MainUdp: Wrote 53 bytes
Endpoint [4]usbUART: got message 11020 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [5]MainUdp: got message 11020 to -1/-1 from 1/1
	Known components:
		252/1
Endpoint [5] accepted message 11020 to -1/-1 from 1/1
UDP [5]MainUdp: Wrote 16 bytes
> UDP [5]MainUdp: Got 21 bytes
Endpoint [4]usbUART: got message 0 to -1/-1 from 252/1
	Known components:
		1/1
Endpoint [4] accepted message 0 to -1/-1 from 252/1
UART [4]usbUART: Wrote 21 bytes
Endpoint [5]MainUdp: got message 0 to -1/-1 from 252/1
	Known components:
		252/1

```

I think there is something funky going on with  sys/comp id matching with mavlink-router.

lucasdemarchi, 2022-12-18:
```
Endpoint [5]MainUdp: got message 0 to -1/-1 from 252/1
	Known components:
		252/1
		1/1
```

It seems apmplanner is connected to both the TCP and the UDP endpoint, causing a loop between the endpoints.
That is probably because APM planner is listening to that port and when we start mavlink-router we are broadcasting to that port.  Either disable the UDP in APM planner or if there isn't such an option, switch to another port so APM planner doesn't  try to use both endpoints.

kaangoksal, 2022-12-18:
@lucasdemarchi  

I turned off the udp endpoint 
```
pi@ras-pilot:~/mavlink-router/build/src $ cat /etc/mavlink-router/main.conf 

[UartEndpoint usbUART]
Device = /dev/ttyACM0  
Baud = 115200
MavlinkDialect = auto
#AllowSrcSysIn = 1

#[UartEndpoint pinUART]
#Device = /dev/ttyS0
#Baud = 57600

#[UdpEndpoint MainUdp]
#mode = Normal
#Address = 10.0.0.167
#Port = 14550
#AllowSrcSysIn = 252
pi@ras-pilot:~/mavlink-router/build/src $ 
```

here is the result 

```
Message 168 to unknown sysid/compid: -1/-1
Endpoint [4]usbUART: got message 136 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [7]dynamic: got message 136 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 136 to unknown sysid/compid: -1/-1
> UART [4]usbUART: Got 61 bytes
Endpoint [4]usbUART: got message 193 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [7]dynamic: got message 193 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 193 to unknown sysid/compid: -1/-1
Endpoint [4]usbUART: got message 241 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [7]dynamic: got message 241 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 241 to unknown sysid/compid: -1/-1
> UART [4]usbUART: Got 69 bytes
Endpoint [4]usbUART: got message 147 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [7]dynamic: got message 147 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 147 to unknown sysid/compid: -1/-1
Endpoint [4]usbUART: got message 11020 to -1/-1 from 1/1
	Known components:
		1/1
Endpoint [7]dynamic: got message 11020 to -1/-1 from 1/1
	Known components:
		252/1
		1/1
Message 11020 to unknown sysid/compid: -1/-1
```

still the same

jbeyerstedt, 2022-12-19:
Are you sure, that the FMU isn't connected though another link to APM and that APM doesn't forward messages between links? Because it's quite unusual, that the endpoint to APM is receiving messages from 1/1.
You could try to use filters when connecting to APM via an UDP endpoint, but that's not actually solving the cause of the problem.

As mavlink-router rejects to send messages where the source sys ID and component ID matches any of the "known components" on this endpoint to prevent loops, it's bad, if 1/1 is present on all endpoints. System and component ID -1/-1 is used for "broadcast" MAVLink messages (where no target IDs are present). Only some messages have a target address (e.g. parameter request and parameter write) as telemetry is usually broadcast to everybody.

kaangoksal, 2022-12-20:
so what this error message is saying that, TCP link has two components BEHIND it, 252/1 and 1/1, that would only happen if APM sent a message as 1/1 to the router? 

my setup is pretty simple, just APM Planner in a seperate computer and raspberry pi ---> mateksys ardupilot on the same network.

closed 2024-12-10 by jbeyerstedt