mavlink-router/mavlink-router #386: Mavlink-router is no longer defaulting to a TCP V4 Server Port.

SGDucharme, 2023-02-17:
Is this be design?  I could not find any information in the README.md file that this behavior was changed from TCP V4 to TCP V6.

My usage of Mavlink-router needs to use TCP V4.

We have modified Mainloop::tcp_open to use TCP V4 instead of TCP V6 for our use.

Would it be better to add a new command line argument that selects which version of TCP is used for the default port?

lucasdemarchi, 2023-02-17:
it should be listening to both

lucasdemarchi, 2023-02-17:
If it's not. After starting mavlink-proxy can you check all the ports it's listening to with something like netstat?  If it's not listening on TCPv4, can you send a patch to fix it?

jbeyerstedt, 2023-02-17:
I definitely have a mavlink-router instance (code version from about 2022-08) running which is reached via IPv4 (VPN) on it's TCP port. So I would be surprised, if it doesn't work any more.

If you open an IPv6 socket in linux, it will also listen to IPv6 on that port by default. It might be confusing, that `netstat -tlpn` only lists the port as "tcp6". (You can explicitly disable this dual-protocol behavior when creating the socket, but that's not something we did.)

SGDucharme, 2023-02-20:
I did some more testing, and it indeed does work.

My confusion was that netstat did not list the TCP v4 listening socket, but only the V6 address.

Thanks for the quick answers.

closed 2023-02-20 by lucasdemarchi