mavlink-router/mavlink-router #388: Autopilot to multipath data link with SATCOM - How to filter and configure?

relaxibus, 2023-03-10:
Hi, I'm working on a new modular multi-wan communication gateway for drones that includes RF (IP), multi-LTE and SATCOM links. The main issue is the SATCOM link, as the max bandwidth available is 5 Kbps, so only selected MAVLink messages at a specific frequency should be sent over this link. Over the LTE we will send more messages and all messages over the RF link.
Attached the block diagrams. The topology is composed by the Remote Vehicle part, the Cloud Aggregator and the Ground Station part.

Question: I have some questions about the mavlink-router works:
1) autopilot connected via UART: will the mavlink-router request to the AP to send his heartbeat and broadcast it to the configured endpoints LTE, RF and SAT? Can the mavlink router automatic request specific messages to the AP such as mavp2p?
2) autopilot connected via serial-to-UDP gateway: I suppose the gateway must be configured in UDP client mode as the mavlink router IN-UDP endpoint is in server mode?
3) how can the autopilot be "stimulated" to broadcast all the telemetry data streams to all endpoints and to use "AllowMsgIdOut" on each endpoint to filter out just the relevant messages? Must this be done on each endpoint or can I create a "stimulation endpoint" where I send message request to the autopilot and he will broadcast back to all endpoints where filtering takes place?

[image]
[image]
[image]
[image]

jbeyerstedt, 2023-03-10:
I only know PX4 and have never used ArduPilot. As far as I know, this stream request feature of ArduPilot is not supported by mavlink-router. So you need to statically configure a telemetry link to mavlink-router which then could forward it to multiple endpoints. It "only" routes (and filters) what's coming in and does not interact with the autopilot (besides this logging features). The autopilot only sees it's link to mavlink-router and that there's a GCS behind it (if it has sent a heartbeat), but not how it is connected of if it's connected via multiple links.

So from my point of view, you have two options for a link with a reduced bandwidth:
1. Configure the SATCOM link as a separate telemetry port on your autopilot besides the telemetry data going to mavlink-router on the UAV side (for LTE and RF). On the ground-side, you can use a single mavlink-router to aggregate and de-duplicate all data links.
2. Only use one connection to the autopilot and use mavlink-router to filter our unwanted message IDs on the SATCOM endpoint (as proposed in your diagrams). This wouldn't reduce the update rate of individual messages though.

Remark on mavlink-router routing: To prevent message loops, a message will not be transmitted on an endpoint, if the message's source address is in the list of "connected systems" on this endpoint. This list contains all source addresses of incoming messages on this endpoint.
To synchronize this list over multiple endpoints, you need to put endpoint of parallel data links in one "group". Otherwise the ground will forward telemetry coming in on one endpoint (e.g. RF) to your other links (e.g. LTE) as well instead of just your GCS.

Sorry for not answering your questions directly, but maybe this helps more by providing the background information.

jbeyerstedt, 2023-03-10:
As long as PX4 can connect to mavlink-router (which is always the case via UART and to a UDP server mode port or TCP server port), it'll start sending heartbeats after boot. IIRC it might stop after a while, if it hasn't received a heartbeat from another system though (but QGroundControl will also not just send heartbeats, if no drone was connected).
The amount of telemetry data, which is sent, is usually configured on a per-port basis. There are a few different [modes](https://docs.px4.io/main/en/peripherals/mavlink_peripherals.html#MAV_X_MODE), which define the message update rate and which message IDs are sent, e.g. "onboard" has high-frequency sensor data while normal telemetry does not.

When taking a look at [the source code](https://github.com/PX4/PX4-Autopilot/blob/9be8f81d754d71a1357e739a21cac2c3d9be1acb/src/modules/mavlink/mavlink_receiver.cpp#L517), it seems like `MAV_CMD_SET_MESSAGE_INTERVAL` is supported. But please ask the PX4 forum for details on that.

relaxibus, 2023-03-10:
I See... this also means that PX4 uses "MAVLINK_MODE" to get telemetry streams and ardupilot "MAV_CMD_SET_MESSAGE_INTERVAL". Cool, MAVLink is a standard but everyone cook its own soup how to use it to make interoperability between platforms very easy... Anyway, I have still this fundamental open question regarding the mavlink router:

My idea was to create a new client endpoint, lets call it Message_Trigger_EP. This endpoint is routed to the localhost that is in charge to issue the "MAV_CMD_SET_MESSAGE_INTERVAL" (but to who? autopilot SyS-ID?). After issued the command, will the autopilot broadcast the requested messages to ALL available endpoints or just to the Message_Trigger_EP?

jbeyerstedt, 2023-03-10:
>  After issued the command, will the autopilot broadcast the requested messages to ALL available endpoints or just to the Message_Trigger_EP?

If the autopilot is only connected to mavlink-router via one link, it does not know from which of the endpoints at mavlink-route the MAVLink message came. MAVLink messages are not like multiple, separate TCP connections.

[These](https://github.com/mavlink-router/mavlink-router#message-routing) are the routing rules. As normal telemetry data has no target address, those are always a broadcast. mavlink-router will route broadcast messages to all endpoints at which the source system is not connected. Only if there's a target address, the message will only be routed to the endpoint at which the targeted system is connected.

For me, mavlink-router works more like an ethernet switch than an IP router as we only know what address is connected on an endpoint (if it has sent data to us once) instead of having a routing table. (It doesn't fall back to broadcast though, if a target address is not known. But most MAVLink messages are broadcast anyways.)

closed 2023-03-14 by relaxibus