mavlink-router/mavlink-router #393: Error binding IPv4 socket

cyb3rL1ght, 2023-04-16:
_Good day!_
I use this connection scheme:

[image]
[image]

however, the mavlink-router can't see my device! 
but if you run `./mavsdk_server udp://15.15.15.15:14550`, the connection will be established!
IP_TABLES and UFW are disabled

I reinstalled the system several times and reinstalled the mavlink-router...but the result is the same!

can you tell me which way to dig???

jbeyerstedt, 2023-04-17:
So to summarize: Your companion computer is running mavlink-router and is connected via ethernet to the "Pix 6x" FMU. The FMU has IP address `15.15.15.15` and the companion has `15.15.15.1`. On a second network interface, the companion computer can reach other devices running QGC.

First of all, you should use IPv4 addresses from https://datatracker.ietf.org/doc/html/rfc1918 in local networks, not any random address. With your configuration, the companion computer will not be able to reach some parts of the internet, which are using the `15.15.15.0/24` address range as this range is routed to the `enx00...` interface.

But the actual issue is, that the `address` field in UDP "server" mode specifies the listing address, not the target address. As "15.15.15.15" isn't owned by any network interface of the device mavlink-router is running on, the operating system can't open an UDP port on that address.
Usually I would let the FMU initiate the MAVLink communication with UDP endpoints, so "server" mode would be OK. In that case, the address needs to be "15.15.15.1". If you want to switch to "normal" mode (which might be what mavsdk_server is doing), the address needs to stay at the address of your FMU.

Additionally, I would recommend to add all endpoints in the config file instead of mixing it with CLI arguments. This would also make it easier to run mavlink-router via systemd.

(Note: "server" mode was named "eavesdropping" in previous versions.)

cyb3rL1ght, 2023-04-17:
> Итак, подведем итоги: ваш компьютер-компаньон работает под управлением mavlink-router и подключен через Ethernet к FMU «Pix 6x». FMU имеет IP-адрес, а компаньон имеет . На втором сетевом интерфейсе компьютер-компаньон может подключаться к другим устройствам, работающим под управлением QGC.`15.15.15.15``15.15.15.1`
> 
> Прежде всего, вы должны использовать IPv4-адреса от https://datatracker.ietf.org/doc/html/rfc1918 в локальных сетях, а не какой-либо случайный адрес. С вашей конфигурацией компьютер-компаньон не сможет получить доступ к некоторым частям Интернета, которые используют диапазон адресов, поскольку этот диапазон направляется к интерфейсу.`15.15.15.0/24``enx00...`
> 
> Но на самом деле проблема заключается в том, что в поле в режиме UDP «сервер» указан адрес листинга, а не целевой адрес. Поскольку «15.15.15.15» не принадлежит какому-либо сетевому интерфейсу устройства, на котором работает mavlink-router, операционная система не может открыть UDP-порт по этому адресу. Обычно я позволяю FMU инициировать связь MAVLink с конечными точками UDP, поэтому «серверный» режим будет в порядке. В этом случае адрес должен быть «15.15.15.1». Если вы хотите переключиться в «обычный» режим (что может быть тем, что mavsdk_server делает), адрес должен оставаться по адресу вашего FMU.`address`
> 
> Кроме того, я бы рекомендовал добавить все конечные точки в конфигурационный файл вместо того, чтобы смешивать его с аргументами CLI. Это также облегчит запуск mavlink-router через systemd.
> 
> (Примечание: в предыдущих версиях «серверный» режим назывался «подслушивание».)

**I swear it worked!**
I had to change the network from 15.15.15.0/24 to 10.10.1.0/24 and everything worked right away!!! 

that's great! 
I also added all UdpEndpoints to main.conf at once

closed 2023-04-22 by cyb3rL1ght