mavlink-router/mavlink-router #394: Full Duplex routing between Serial and UDP end points

Mas313, 2023-05-11:
Hello mavlink-router developers. I am new using mavlink router. I want auto pilot to receive mavlink RC commands from ground station and also at the same time send telemetry data to ground station.
Does mavlink-router support bi-directional full duplex uart (serial port) and udp socket operation.
Would a serial transmission by auto pilot send data to udp end point and transmission by a socket send data to serial end point?
here is my configuration file

[General]
   TcpServerPort=5760
   ReportStats=false
   MavlinkDialect=common

```
[UartEndpoint serial0]
   Device=/dev/ttyS0
   Baud=38400

[UdpEndpoint local]
   Mode=normal
   Address=127.0.0.1
   Port=14550
```
In one terminal I would use
`mavlink-routerd -e 127.0.0.1:14550`

In second terminal a program in a thread will be writing to UDP end point the RC commands received from ground that will be sent to serial end point and second thread would be reading (Mavlink telemetry) data from udp end point received through same serial port.
Would my config file serve the mentioned purpose?

jbeyerstedt, 2023-05-12:
All endpoints are able to transmit and receive data "full duplex". (Since there is only one event loop, messages will actually be processed sequentially.)

You just have to keep in mind, that an UDP endpoint always needs one part to "initiate" the "connection" as only the target or the listening port can be provided in either `normal` or `server` mode.  
In `normal` mode, it is expected, that mavlink-router will send the first data to the specified client address and port. After the client has received something, it knows which address and port mavlink-router has and can send data back.  
In `server` mode, it's the other was around. Mavlink-router is listening in the specified port, but doesn't know the address and port of the client.

With your config file, you don't need to specify the endpoint as an CLI argument as well. Mavlink-router will first create all endpoints found in the config file and then add additional ones you specified via the CLI arguments. So your example won't work as you defined `127.0.0.1:14550` twice.
My personal recommendation is to use the config file instead of the CLI way as  it has less implicit assumptions and can be used in conjunction with a systemd unit.

jbeyerstedt, 2024-12-10:
Closing because of no activity. Feel free to comment and re-open, if there are things left to discuss.