mavlink-router/mavlink-router #403: Messages Filters : EndPoint( receiver ,port : 14553) receives all messages from all other EndPoints even throw i used AllowSrcSysIn from specific Endpoint(sender , port: 14552)

abdullahalabboud, 2023-08-17:
i created two python file to send and receive text message using pymavlink (statustext_send()) with this : 

**sender.py** :
```
connection= mavutil.mavlink_connection("udp:14552" , source_system=21)
.
.
.
connection.mav.statustext_send(
    mavutil.mavlink.MAV_SEVERITY_INFO, "text message".encode()
)
```

**receiver.py** :

```
connection= mavutil.mavlink_connection("udp:14553" , source_system=22 , source_component=1)
.
.
.
while True : 
     msg = mav.recv_match(blocking=True)
    print("Message from %d %s " % (msg.get_srcSystem(), msg))

```
main.config
```
[General]
   TcpServerPort=5760
   ReportStats=false
   MavlinkDialect=common

[UdpEndpoint local]
   Mode=server
   Address=127.0.0.1
   Port=14550

[UdpEndpoint sender]
   Mode=normal
   Address=127.0.0.1
   Port=14552

[UdpEndpoint receiver]
   Mode=normal
   Address=127.0.0.1
   Port=14553
   AllowSrcSysIn=[21]
```

NOT : I am using SiTL

jbeyerstedt, 2023-08-22:
`AllowSrcSysIn` shall be a comma separated list of values: https://github.com/mavlink-router/mavlink-router/blob/master/examples/config.sample#L18

abdullahalabboud, 2023-08-22:
_# Lists are comma separated._ 
but i can't understand how can do it .... i am trying like this [21] and i tried like this 21,23 and like this [21],[23] for example without any changing in the result .
it stills receives all messages without like the filters of endpoints not set

jbeyerstedt, 2023-08-22:
Sorry, I seem to have read over the important part. You want to limit what's going out at this endpoint, right?

The documentation states:
> AllowSrcSysIn: If set, only allow messages from the listed MAVLink source systems to be received on this endpoint

So this filter affects all messages coming in from the outside at this endpoint. It does not affect messages routed from other endpoints and being sent out at this endpoint.  
So please try `AllowSrcSysOut` instead.

`AllowSrcSysIn=21` or `AllowSrcSysIn=21, 42` should be the right syntax nonetheless.

jbeyerstedt, 2024-12-10:
Closing because of no activity. Feel free to comment and re-open, if there are things left to discuss.