mavlink-router/mavlink-router #411: remote connection using mavlink router

MIKE9708, 2023-11-16:
hi, i was trying to set up a comunication between to machines, one with the SITL, and the other containing just a python script that use MASDK.In order to make the comunication work i wanted to use mavlink router that from what i've read on the PX4 documentation allows to forward messages that would be sent to localhost , to a specific ip and port, now i followed the example for the conf file that px4 provide but it does not work , and i do not understand if i am makinkg some mistake.
Bu the way [this](https://docs.px4.io/main/en/simulation/#use-mavlink-router) is the PX4 docu section where it talks about mavlink router

jbeyerstedt, 2023-11-17:
The wording in the PX4 documentation is a bit misleading. In general mavlink-router can route packets from and to any of its supported endpoints (UART, UDP and TCP), even with multiple ones.
If I remember it correctly, the PX4 SITL script is sending UDP data to localhost, port 14550. So mavlink-router needs to listen for incoming data on that port (that's the "eavesdropping" part) and send it to your other machine (the "normal" mode). And your MAVSDK script needs to listen for UDP data on the configured port obviously (e.g. with `udp://0.0.0.0:14550`).

Regarding the config file: It should still work, but we renamed "eavesdropping" mode to "server" some time ago (keeping the old value as a valid, but deprecated option).
I always recommend using the config file over the CLI options as it's more explicit about the options and therefore easier to understand (in my opinion). And the config file works better with a systemd unit.

If it still doesn't work, please provide the relevant configuration of mavlink-router and your MAVSDK script, as well as a snippet of the mavlink-routerd output when run in verbose mode (add `-v` to the CLI arguments). In verbose mode you should see, if messages were received and where they will be forwarded to.

MIKE9708, 2023-11-17:
Ok thanks for the replay i'll try what you said and i'll let you know

MIKE9708, 2023-11-18:
Ok i tried and it works perfect thank you for the explanation

closed 2023-11-18 by MIKE9708