mavlink-router/mavlink-router #412: How am I supposed to write C++ application that receives data from Mavlink router and sends data to the mavlink network at the same time?

eMrazSVK, 2023-11-27:
I am trying to write C++ app, that would listen to UDP endpoint and then responds with data. The behavior is following:
- I receive data from configured endpoint in my app correctly - I can receive all data and all heartbeats from all devices
- I save address and port from which the message was sent (I check for heartbeat) and then I try to response to that address:port
- Everything works, except from the time I send my own HEARTBEAT, I stop receiving messages which I was receiving befor just right
- I guess this is somehow expected when I read readme (https://github.com/mavlink-router/mavlink-router#endpoints), especially:
> When using any non-unicast IP address, e.g. an IPv4 broadcast or IPv6 local network multicast (ff02::1), messages will be "broadcasted" until somebody sends data back. From then on, UDP packets will only be sent to the specific unicast IP of the answering device.
- But then I am here and I am asking for help of how to accomplish this task since I want to be able to both send and receive mavlink messages using mavlink router? What am I missing? Thank you.

jbeyerstedt, 2023-11-28:
Just to recap your configuration:
- Your application binds to some UDP port, e.g. 14542 on `INADDR_ANY`
- Mavlink-router is configured to send data to port 14542 on the IP address where your application is running (endpoint is set to UDP "normal" mode)

(If not, please share your mavlink-router config and the how you open your UDP port.)

Sending data back to mavlink-router should not stop your endpoint from receiving data unless your MAVLink system ID (and component ID) is the same as what your other system(s) use. The cited section should not apply at all (

For further debugging, you should run mavlink-router with in verbose mode (`-v` CLI flag) and try to observe how messages are routed between your endpoints. It should give you some information, if messages are rejected because they are supposedly routed in a loop (which is determined by the source sys ID/ comp ID).

eMrazSVK, 2023-11-29:
This is my mavlink-router config:
```
[General]
DebugLogLevel = debug

# Main connection to AutoPilot
[UartEndpoint cube]
Device = /dev/ttyTHS0
Baud = 921600

# GCS
[UdpEndpoint qgc]
Mode = Normal
Address = 192.168.55.100
Port = 14550

# CustomApp
[UdpEndpoint CustomApp]
Mode = Normal
Address = 127.0.0.1
Port = 14560
```

Port setup in CustomApp:
```
// receive endpoint initialized like this:
//  receive_endpoint_(
//    boost::asio::ip::udp::endpoint(
//     boost::asio::ip::address::from_string(target_ip), udp_port))

void UDP_Port::start() {
  // Open the socket for the given address family
  receive_socket_.open(receive_endpoint_.protocol());
  // Bind the socket to the given endpoint
  receive_socket_.bind(receive_endpoint_);
  read_thread_ = std::thread(&UDP_Port::read_loop, this);
}

```

And then when I receive first message:
```
void UDP_Port::read_loop() {
  uint8_t tmp_read_buffer[1024]; // Adjust read_buffer_ size as needed
  boost::asio::ip::udp::endpoint sender_endpoint;

  std::function<void(const boost::system::error_code&, std::size_t)> handler;
  handler = [this, &tmp_read_buffer, &sender_endpoint, &handler](
    const boost::system::error_code& ec, std::size_t bytes_recvd) {
    if (!ec && bytes_recvd > 0) {
      if (!write_socket_.is_open()) {
        write_socket_.open(sender_endpoint.protocol());
        write_endpoint_ = sender_endpoint;
        write_thread_ = std::thread(&UDP_Port::write_loop, this);
      }
      std::cout << "ram" << std::endl;
      std::lock_guard<std::mutex> lock(read_mutex_);
      
      for (size_t i = 0; i < bytes_recvd; ++i) {
        read_buffer_.push_back(tmp_read_buffer[i]);
      }
    }
    // Continue reading
    receive_socket_.async_receive_from(boost::asio::buffer(tmp_read_buffer), sender_endpoint, handler);
  };

  receive_socket_.async_receive_from(boost::asio::buffer(tmp_read_buffer), sender_endpoint, handler);

  receive_io_service.run();
}

```
I save `sender_endpoint` and I start write thread to write data to where it came from:
```
void UDP_Port::write_loop() {
  while (true) {
    mavlink_message_t message;
    {
      if (!write_socket_.is_open()) continue;
      if (!write_buffer_.empty()) {
        std::unique_lock<std::mutex> lock(write_mutex_);
        message = write_buffer_.front();
        write_buffer_.pop_front();
      } else {
        std::this_thread::sleep_for(std::chrono::milliseconds(10));
        continue;
      }
    }

    uint8_t buffer[MAVLINK_MAX_PACKET_LEN];
    int length = mavlink_msg_to_send_buffer(buffer, &message);
    write_socket_.send_to(boost::asio::buffer(buffer, length), write_endpoint_);
  }
}
```

I start it with like this: 
```

eMrazSVK, 2023-11-29:
I found the problem. I use two different sockets for receiving and sending the data:
- `write_socket_`
- `receive_socket_`
When I changed this to use just one socket for both operations, it works this way. Thanks for the help though!

eMrazSVK, 2023-11-29:
Sorry, I am reopening this issue, just to ask: @jbeyerstedt would it be valuable if I contributed to the `mavlink-router` with this example of C++ app communicating with mavlink-router (send/receive) as I imagine this is a common use case for people. It also uses more modern approach (std and boost libs).

jbeyerstedt, 2023-11-29:
From my point of view, this repo is not the right place for a full C++ example application as it would need some documentation about the dependencies (as you are using boost) and probably some build system. This is way more than the python scripts in the example directory.

eMrazSVK, 2023-11-29:
Okay, maybe I will post an example repo and then you can refer to it in the readme for example if you would like. I will let you know if that's the case in the near future :)

closed 2023-11-29 by eMrazSVK