mavlink-router/mavlink-router #413: Dynamic endpoint manipulation

eMrazSVK, 2023-12-07:
Hello, 

I am thinking of swarming applications - which are more prevalent everyday. Imagine we have a network with multiple UAV's (mavlink devices) that might connect/disconnect during one mission. Let's say we have a discovery protocol which is running in a background. From my point of view, it would be beneficial if mavlink-router supported adding/removing endpoints during runtime for such application. Is this something you have considered in the past? What about contributing with such feature?

eMrazSVK, 2023-12-08:
Yeah, UART would be useless or it would be too much hassle for very little benefit. I will take a look at #242 PR and maybe I can inspire myself. So to describe my use case:

- I am mainly interested in adding/removing UDP endpoints during runtime.
- The use case is to have swarm of UAV's on one network also with GCS's, custom Mavlink payloads etc.
- These devices might disconnect (which is not that bad) or connect (turn on) after some initialization process after network device discovery.
- So in practice, there would be custom Linux app that would basically do network discovery - sending broadcast messages on each device. So each device identify itself with the others.
- Based on this data, the custom app would be able to add/remove UDP endpoints inside Mavlink router -> so I am thinking of some "daemon" behaviour of mavlink-router where you are able to interact with it during runtime.

Is this clear enough? I might take a look at it, take a look at the mentioned PR and I would love this feature inside mavlink-router, because despite this, it works good for us.

eMrazSVK, 2023-12-11:
I want to be able to communicate with all drones and also drone to drone and I do not want to broadcast everything. The network discovery surely CAN be implemented by broadcasting heartbeat for example, but if I want to communicate with UAV's that respond, I need to add endpoint inside mavlink router to satisfy that ability.

jbeyerstedt, 2023-12-11:
How would the UDP communication be set up? Would you use "normal" mode UDP endpoint at the central mavlink-router or "server" mode?
If "normal" mode: Are you sure, that somebody else will send a "broadcast" MAVLink message (like a heartbeat) first? Otherwise your drone won't be able to send anything to the central server as it doesn't know its UDP port.
If "server" mode: How would the port be communicated to the drone? And why not configure everything beforehand? You need to make sure, that MAVLink system IDs aren't re-used anyways.

Those are the questions I would like to have answered with a slightly more detailed sketch of the concept.

lucasdemarchi, 2023-12-11:
I was thinking on all endpoints in "normal" mode and either make mavlink-router itself send a heartbeat or start a client together with mavlink-router to serve that purpose

eMrazSVK, 2023-12-11:
Ok, so I'll try to provide a more systematic concept of how I imagine it:
- the base for this is the ability to add UDP endpoints at runtime, so exactly the same behavior like the mavlink-router works now.
- "the" discovery app is meant to be outside of mavlink-router and would do this exactly:
  - each UAV is on the same subnet -> we plan to utilize mav ID's for IP address setting too, so the IP looks like this: X.Y.Z.SYS_ID
  - the discovery app sends (broadcasts) kind of heartbeat, does not need to be mavlink heartbeat. 
  - the discovery app also receives those "heartbeat" broadcast msgs. 
  - **According to the received messages, UDP endpoints in mavlink-router are added** during runtime. I think `normal` mode is okay for this.
  - This way, the UAV with added endpoint like this gained ability/knowledge of WHAT mavlink system id is on WHICH IP Address (endpoint) and the UAV is now able to communicate with another UAV.
-  This kind of communication works good if I find IP of each device/UAV in the network, but I just want to do this dynamically, at runtime.

Does this make sense?

jbeyerstedt, 2023-12-11:
Now I'm more confused than before: Do you want to run one mavlink-router instance with dynamic endpoints on each drone or do you have a central "server" with on mavlink-router instance though which all mavlink communication is running?

lucasdemarchi, 2023-12-11:
> I want to be able to communicate with all drones and also drone to drone and I do not want to broadcast everything. The network discovery surely CAN be implemented by broadcasting heartbeat for [...]

a heartbeat is part of any flight stack on the drone and GCSs/apps. What I'm saying is that you don't need the discovery to be "out of band"  if there's a heartbeat from mavlink-router that allows the drones/GCS to discover mavlink-router itself.  When they discover mavlink-router they will start sending there data there and that will cause the endpoint addition at runtime. With that endpoint added, mavlink-router will route any message from that endpoint to the other possible endpoints: if it's a mavlink-broadcast, to all other endpoints, or if it's to a specific system-id, only to the endpoint where that system is.

eMrazSVK, 2023-12-12:
@jbeyerstedt I want mavlink-router running on each drone/GCS any Mavlink device really.

@lucasdemarchi okay, the "discovery app" was only an idea that was not really thought through. So yeah I agree with your suggestion regarding discovery. Now, if I understand it right, I still need to add a runtime endpoint addition to the mavlink-router for this whole concept to work, right? And also heartbeat sending from mavlink router itself. I think it's quite clever to do this, I was just not sure if this discovery behavior is in scope of mavlink-router, but it makes sense.