mavlink-router/mavlink-router #416: Allow using hostname instead of IP address for endpoints

bringert, 2024-02-05:
In v3, trying to use a hostname instead of an IP address in the value for the --endpoint argument fails:

```
$ ./mavlink-routerd-glibc-aarch64 -e mycomputer:14550 /dev/serial0:921600
mavlink-router version v3
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
UdpEndpoint CLI: Invalid IP address mycomputer
```

It would be nice to be able to use a hostname instead of having to hardcode an IP address or run an extra command to do the lookup.

bringert, 2024-02-05:
Workaround:

```
$ ./mavlink-routerd-glibc-aarch64 -e $(getent ahostsv4 mycomputer | awk '{ print $1 }' | head -1):14550 /dev/serial0:921600
```

lucasdemarchi, 2024-02-05:
yep, resolving the name when we are starting mavlink-router should be doable. Could you send a patch for that?

bringert, 2024-04-11:
I've got a first working version here: https://github.com/mavlink-router/mavlink-router/commit/76ca7e348a1a4666b796aac44cc5c3c6e46b63a5

It currently makes the tests fail, since we no longer validate IP addresses. I should probably change the validation function to do hostname lookup too.

JasonBrauer, 2024-07-31:
@lucasdemarchi Are there plans to get the hostname lookup integrated into master? We use the config file, so it would be especially useful in that case.