mavlink-router/mavlink-router #418: Raspbian OS lite (64bit) getting stucked after "ninja -C build" command

dr-electron96, 2024-03-18:
System configuration:
Raspberry pi zero 2w
Raspbian OS lite 64bit
8gb sd card.
using terminal from putty via WiFi SSH.

Issue:
when I execute "ninja -C build" command the most of the time terminal freezes on ninja: Entering directory 'build' [1/22] Compiling c++ object src/mavlink-routerd.p/binlog.cpp" or gets stuck at "[4/25] Compiling c++ object src/mavlink-routerd.p/binlog.o".

jbeyerstedt, 2024-03-18:
Looking at previous issues related to compiling on an RPi Zero (like #349), I would suspect that you run out of RAM. If so, you basically have three options:
- Decrease the number of parallel compile jobs to 1 by adding `-j 1` to the ninja command
- Increase swap size
- Cross-compile outside the Raspberry Pi Zero and copy the result over

dr-electron96, 2024-03-21:
@jbeyerstedt I greatly appreciate your speedy Input, my issue is now resolved.

closed 2024-03-21 by dr-electron96