mavlink-router/mavlink-router #419: ninja: error: unknown target 'clang-format'

bringert, 2024-04-10:
The README says that coding style changes can be done automatically by running `ninja -C build clang-format`. But when I try I get `ninja: error: unknown target 'clang-format'`.

```
# meson setup build .
The Meson build system
Version: 1.0.1
Source dir: /mavlink-router
Build dir: /mavlink-router/build
Build type: native build
Project name: mavlink-router
Project version: 3
C compiler for the host machine: cc (gcc 12.2.0 "cc (Debian 12.2.0-14) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14) 12.2.0")
C++ linker for the host machine: c++ ld.bfd 2.40
Host machine cpu family: aarch64
Host machine cpu: aarch64
Library m found: YES
Library rt found: YES
Run-time dependency threads found: YES
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Run-time dependency systemd found: YES 252
Run-time dependency GTest found: YES (building self)
Compiler for C supports arguments -Wno-inline: YES 
Compiler for C supports arguments -Wundef: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wlogical-op: YES 
Compiler for C supports arguments -Wsign-compare: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wfloat-equal: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wredundant-decls: YES 
Compiler for C supports arguments -Wmissing-declarations: YES 
Compiler for C supports arguments -Wmissing-noreturn: YES 
Compiler for C supports arguments -Wshadow: YES 
Compiler for C supports arguments -Wendif-labels: YES 
Compiler for C supports arguments -Wstrict-aliasing=3: YES 
Compiler for C supports arguments -Wwrite-strings: YES 
Compiler for C supports arguments -Wno-long-long: YES 
Compiler for C supports arguments -Wno-overlength-strings: YES 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-unused-result: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wchar-subscripts: YES 
Compiler for C supports arguments -Wtype-limits: YES 
Compiler for C supports arguments -Wuninitialized: YES 
Compiler for C supports arguments -Waddress-of-packed-member: YES 
Compiler for C supports arguments -Wno-inline: YES (cached)
Compiler for C supports arguments -Wundef: YES (cached)
Compiler for C supports arguments -Wformat=2: YES (cached)
Compiler for C supports arguments -Wlogical-op: YES (cached)
Compiler for C supports arguments -Wsign-compare: YES (cached)
Compiler for C supports arguments -Wformat-security: YES (cached)
Compiler for C supports arguments -Wmissing-include-dirs: YES (cached)
Compiler for C supports arguments -Wformat-nonliteral: YES (cached)
Compiler for C supports arguments -Wpointer-arith: YES (cached)
Compiler for C supports arguments -Winit-self: YES (cached)
Compiler for C supports arguments -Wfloat-equal: YES (cached)
Compiler for C supports arguments -Wredundant-decls: YES (cached)
Compiler for C supports arguments -Wmissing-declarations: YES (cached)
Compiler for C supports arguments -Wmissing-noreturn: YES (cached)
Compiler for C supports arguments -Wshadow: YES (cached)
Compiler for C supports arguments -Wendif-labels: YES (cached)
Compiler for C supports arguments -Wstrict-aliasing=3: YES (cached)
Compiler for C supports arguments -Wwrite-strings: YES (cached)
Compiler for C supports arguments -Wno-long-long: YES (cached)
Compiler for C supports arguments -Wno-overlength-strings: YES (cached)
Compiler for C supports arguments -Wno-unused-parameter: YES (cached)
Compiler for C supports arguments -Wno-missing-field-initializers: YES (cached)
Compiler for C supports arguments -Wno-unused-result: YES (cached)
Compiler for C supports arguments -Wchar-subscripts: YES (cached)
Compiler for C supports arguments -Wtype-limits: YES (cached)
Compiler for C supports arguments -Wuninitialized: YES (cached)
Compiler for C++ supports arguments -Waddress-of-packed-member: YES 
Header "aio.h" has symbol "aio_init" : YES 
Configuring config.h using configuration
Configuring mavlink-router.service using configuration
Program run-clang-tidy found: YES (/usr/bin/run-clang-tidy)
Build targets in project: 10

Found ninja-1.11.1 at /usr/bin/ninja

root@45499d0b1668:/mavlink-router# ninja -C build
... (omitted lots of googletest warnings)...

# ninja -C build clang-format
ninja: Entering directory `build'
ninja: error: unknown target 'clang-format'
```

Because my host machine is a Mac, I'm doing the build in a Docker container produced by this Dockerfile:

```
FROM debian:stable

RUN apt-get update
RUN apt-get install -y git build-essential autoconf automake meson ninja-build pkg-config gcc g++ systemd
RUN apt-get install -y googletest clang-tidy

WORKDIR mavlink-router 
```

I then create a Docker container with:
```
$ sudo docker run -ti --mount type=bind,source="$(pwd)",target=/mavlink-router --name dev-mavlink-router dev-mavlink-router
```

lucasdemarchi, 2024-04-10:
you probably need to install clang-format in your containers? https://packages.debian.org/search?keywords=clang-format

bringert, 2024-04-11:
Thanks, yes, installing `clang-format` and rerunning `meson setup build .` did the trick. 

I had originally installed `clang-tidy` but not `clang-format`, since that's what `meson setup build .` seems to want: 
```
Program run-clang-tidy found: YES (/usr/bin/run-clang-tidy)
```

Maybe the README should be updated with the `clang-format` dependency? And maybe `meson.build` should check for `clang-format` too?

lucasdemarchi, 2024-04-11:
can you send a patch with those changes?