mavlink-router/mavlink-router #424: Can't use nuttx shell via TCP from Mavlink-rounter PX4

aminballoon, 2024-05-23:
I can't use Nuttx shell via UDP/TCP from Mavlink-rounter.
this is my config 
``` 
[General]
#Mavlink-router serves on this TCP port
TcpServerPort=5760
ReportStats=false
MavlinkDialect=auto

[UartEndpoint FCUSB]
Device = /dev/ttyACM0
Baud = 2000000

[UdpEndpoint GCS]
Mode = Server
Address = 0.0.0.0
Port = 14550

[UdpEndpoint MAVSDK]
Mode = Server
Address = 0.0.0.0
Port = 14540
```

I have try to start mavlink shell via 14540,14550 and TCP 5760 but it can't shell .

jbeyerstedt, 2024-12-10:
Sorry for the late reply, but I can't really tell what's wrong with your config.

What are you using to access the Nuttx shell?

aminballoon, 2025-02-08:
My USB to TTL Chip got problem. Replace to another one it's worked. Thank you

closed 2025-02-08 by aminballoon