mavlink-router/mavlink-router #428: Incomplete messages issue

vanhphm, 2024-08-01:
1. Pixhawk connected with raspi through ttyS0 port
2. My raspi connected to wifi and transmit information to AWS cloud
3. My laptop running Qgroundcontrol and also connect with the AWS server
    My QGC use Mavlink-router 1 (telem 1) baudrate 57600
5. I use this command line: `mavlink-routerd -p 127.0.0.1:5760 /dev/ttyS0:57600`
Then i got this reply: 
```
mavlink-router version v4
Error while trying to write serial port latency on /dev/ttyS0: Operation not permitted
Opened UART [4]alpha: /dev/ttyS0
UART [4]alpha: speed = 57600
Error while trying to write serial port latency on /dev/ttyS0: Operation not permitted
Opened UART [5]CLI: /dev/ttyS0
UART [5]CLI: speed = 57600
Opened UDP Client [6]charlie: 127.0.0.1:5760
Opened UDP Server [8]bravo: 0.0.0.0:5760
9 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
10 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
Opened TCP Server [11] [::]:5760
Opened TCP Client [13]delta: 127.0.0.1:5760
TCP [9]dynamic: Connection accepted
Opened TCP Client [14]CLI: 127.0.0.1:5760
UART Endpoint [5]CLI: 94 incomplete messages in the last 5 seconds
TCP [10]dynamic: Connection accepted
UART Endpoint [4]alpha: 57 incomplete messages in the last 5 seconds
UART Endpoint [4]alpha: 98 incomplete messages in the last 5 seconds
TCP [15]dynamic: Connection accepted
UART Endpoint [4]alpha: 141 incomplete messages in the last 5 seconds
UART Endpoint [5]CLI: 2 incomplete messages in the last 5 second

Thanks in advance 
Sorry for any misinformation, im quite new to raspi and pixhawk.

jbeyerstedt, 2024-08-01:
I'm quite curious how your config file (`/etc/mavlink-router/main.conf`) looks like. Because it seems like you also have some configuration in there, which leads to two endpoints on the same UART device and doesn't work.

(I really recommend to configure everything in the config file. This way you can use the provided systemd unit file as it is to let mavlink-router run as a service.)

vanhphm, 2024-08-02:
> I'm quite curious how your config file (`/etc/mavlink-router/main.conf`) looks like. Because it seems like you also have some configuration in there, which leads to two endpoints on the same UART [...]

```
[UartEndpoint alpha]
Device = /dev/ttyS0
Baud = 57600
[UdpEndpoint bravo]
Mode = Server
Address = 0.0.0.0
Port = 5760
[UdpEndpoint charlie]
Mode = Normal
Address = 127.0.0.1
Port = 5760
[TcpEndpoint delta]
Address = 127.0.0.1
Port = 5760
```

Here my etc/mavlink-router/main.conf
Thanks for your help

jbeyerstedt, 2024-08-05:
Two points:
1. If you have a UART endpoint in the config file you can't use the same device in the CLI options
2. Your UDP and TCP endpoints don't make sense. You have an UDP endpoint (bravo) listening on 0.0.0.0:5760, but also one (charlie) sending to this port via localhost (127.0.0.1:5760). So this is your first message loop. But you also try to open a TCP connection to port 5760 where UDP endpoint "bravo" is listening.

GuiliangZ, 2024-08-05:
We encountered the same issue. We established the connection between the QGC and the RPi through the default TCP port, but no data was transmitted. In our case, it was because we were using RPi zero, and it didn't have enough RAM or other weird issues, but when  we switched to raspberry pi 4 and did everything again, it worked, and QGC and RPi4 could connect seamlessly with MAVlink data transmitted between the two.

vanhphm, 2024-08-07:
> Two points:
> 
> 1. If you have a UART endpoint in the config file you can't use the same device in the CLI options
> 2. Your UDP and TCP endpoints don't make sense. You have an UDP endpoint (bravo) listening on 0.0.0.0:5760, but also one (charlie) sending to this port via localhost (127.0.0.1:5760). So this is your first message loop. But you also try to open a TCP connection to port 5760 where UDP endpoint "bravo" is listening.
> 
> I would recommend to have a config file like this (and not use any endpoints in the CLI params):
> 
> ```
> [UartEndpoint pixhawk]
> Device = /dev/ttyS0
> Baud = 57600
> 
> [TcpEndpoint aws]
> Address = <ip of AWS endpoint>
> Port = <port of AWS endpoint>
> ```
> 
> Or do you have some "bridge" application on your Raspi? Then use a TCP or UDP endpoint to send to that code and it will upload the data to AWS.

My aws ip address require an aws.pem key to authentical. Because of that i can't access to our sever. Therefore to access sever i have to ssh to AWS server both my Laptop and the raspberrypi. So how should i configurate my main.conf. I tried to use this config 
`[UartEndpoint pixhawk]
Device = /dev/ttyS0
Baud = 57600
[TcpEndpoint aws]
Address = 127.0.0.1
Port = 5760
`
here is the reply:
`dronepi@raspberrypi:/etc/mavlink-router $ sudo mavlink-routerd
mavlink-router version v4
Opened UART [4]pixhawk: /dev/ttyS0
UART [4]pixhawk: speed = 57600
5 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
Opened TCP Server [6] [::]:5760
Opened TCP Client [8]aws: 127.0.0.1:5760
TCP [5]dynamic: Connection accepted
6 messages to unknown endpoints in the last 5 seconds
12 messages to unknown endpoints in the last 5 seconds
TCP [9]dynamic: Connection accepted
2 messages to unknown endpoints in the last 5 seconds
`
I see TCP [9]dynamic: Connection accepted but i can't see my qgc stay disconnected and can't received any data.
Thanks for your help

vanhphm, 2024-08-07:
> We encountered the same issue. We established the connection between the QGC and the RPi through the default TCP port, but no data was transmitted. In our case, it was because we were using RPi [...]
What was your setup in raspi and pc, i can use dev/TTYACM0 and also have a good connection between QGC and my Pi but my teacher require me to use dev/TTYS0 and im really struggle with it.
Really appreciate your replied