mavlink-router/mavlink-router #429: Route mavlink packets to the cloud via LTE/4G error: messages to unknown endpoints in the last 5 seconds

huytd-pi, 2024-08-05:
Hello, I use mavlink-router to route mavlink packets to cloud ecs2 aws (free) via LTE/4G, I use raspi 3 mode B+ companion computer.
**When connected usb raspi 3 to pixhawk 4 via usb port, no error messages to unknown endpoints in the last**

> mavlink-routerd -p 127.0.0.1:5760 /dev/ttyACM0
mavlink-router version v4
Opened UART [4]CLI: /dev/ttyACM0
UART [4]CLI: speed = 115200
5 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
6 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
Opened TCP Server [7] [::]:5760
Opened TCP Client [9]delta: 127.0.0.1:5760
TCP [5]dynamic: Connection accepted
Opened TCP Client [10]CLI: 127.0.0.1:5760
TCP [6]dynamic: Connection accepted

**connection via uart on raspi 3 via uart pin 14,15 and pixhawk 4 via telemetry port failed.**

> mavlink-routerd -p 127.0.0.1:5760 /dev/ttyS0:115200
mavlink-router version v4
Error while trying to write serial port latency on /dev/ttyS0: Operation not permitted
Opened UART [4]CLI: /dev/ttyS0
UART [4]CLI: speed = 115200
5 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
6 Error connecting to 127.0.0.1:5760 (Connection refused)
Could not open 127.0.0.1:5760, re-trying every 5 sec
Opened TCP Server [7] [::]:5760
Opened TCP Client [9]delta: 127.0.0.1:5760
TCP [5]dynamic: Connection accepted
Opened TCP Client [10]CLI: 127.0.0.1:5760
TCP [6]dynamic: Connection accepted
17 messages to unknown endpoints in the last 5 seconds
8 messages to unknown endpoints in the last 5 seconds
16 messages to unknown endpoints in the last 5 seconds
17 messages to unknown endpoints in the last 5 seconds
24 messages to unknown endpoints in the last 5 seconds
8 messages to unknown endpoints in the last 5 seconds
1 messages to unknown endpoints in the last 5 seconds
8 messages to unknown endpoints in the last 5 seconds
10 messages to unknown endpoints in the last 5 seconds
26 messages to unknown endpoints in the last 5 seconds

jbeyerstedt, 2024-08-05:
Please don't mix endpoints on the CLI with ones in the config file. It really gets confusing fast.

Second: Where is your connection to AWS? I only see connections to localhost (which actually loop back to the TCP "server" port of mavlink-router itself).

jbeyerstedt, 2024-12-10:
Closing because of no activity. Feel free to comment and re-open, if there are things left to discuss.

closed 2024-12-10 by jbeyerstedt