mavlink-router/mavlink-router #434: 1 messages to unknown endpoints in the last 5 seconds

1776340526, 2024-09-05:
I wanted to use the mavlink-router to connect px4 and qgc. I connected the px4 to /dev/ttyACM of the Raspberry PI and configured port 1.92.***.***, 5760 for forwarding to the public network. The following error is now reported.
[image]
[image]

jbeyerstedt, 2024-09-05:
First a few questions regarding your config:
- What is listening on port 5760 at 1.92.127.128? As far as I know QGC is only listening for UDP traffic on port 14550
- Who is connecting to the Raspberry Pi on port 5760? mavlink-router has a TCP server at that port if not disabled (see [examples/config.sample](https://github.com/mavlink-router/mavlink-router/blob/master/examples/config.sample#L49))

1776340526, 2024-09-06:
> 首先，关于您的配置有几个问题：
> 
> * 在 1.92.127.128 的端口 5760 上侦听什么？据我所知，QGC 仅侦听端口 14550 上的 UDP 流量
> * 谁在端口 5760 上连接到 Raspberry Pi？如果未禁用，mavlink-router 在该端口上有一个 TCP 服务器（参见 [examples/config.sample](https://github.com/mavlink-router/mavlink-router/blob/master/examples/config.sample#L49))

Port 5760 is the port that I forward to the public network, and my QGC has set tcp to connect to port 5760 of the public network.
[image]
Did I set port 5760 again? My 5760 was forwarded to the public network, so I set port 5760 like this.

1776340526, 2024-09-06:
> First a few questions regarding your config:
>
> * What is listening on port 5760 at 1.92.127.128? As far as I know QGC is only listening for UDP traffic on port 14550
> * Who is connecting to the [...]

I tried udp again in ubuntu and it also got an error. Below is a screenshot of the configuration and errors.
[image]
[image]

1776340526, 2024-09-06:
> First a few questions regarding your config:
>
> * What is listening on port 5760 at 1.92.127.128? As far as I know QGC is only listening for UDP traffic on port 14550
> * Who is connecting to the [...]

I think I found the problem. Since the Raspberry PI's port 5760 has been forwarded to the public network, I don't need to do any other configuration.
[image]
[image]