mavlink-router/mavlink-router #435: How can I make connection with UDPCI like MP?

emogooo, 2024-10-18:
As you can see in the image below, I am able to connect to the FCU in MissionPlanner using the address and port `(14000)` I entered via UDPCI. I want to make the same connection in Ubuntu using mavlink-router. However, I am unable to achieve this. Why might that be? My config file is as shown below.

[image]

My Config File:

```
[General]
Log=/mnt/ep
LogMode=always
TcpServerPort=0

[UdpEndpoint listener]
Mode = Normal
Address = 224.1.2.34
Port = 14000
```

jbeyerstedt, 2024-10-18:
Your config only has one endpoint which sends data to `224.1.2.34:14000`, if there is something coming in from another endpoint. But since you don't have any other endpoints, nothing will happen at all.

You need to specify a second endpoint and connect a client which starts to MAVLink data on its own. Otherwise no UDP "connection" will be made and the service on 224.1.2.34 won't be able to send data to you.

emogooo, 2024-10-19:
Thanks for reply. But i worked with config at below. It didn't work too.

```
[General]
Log=/mnt/ep
LogMode=always
TcpServerPort=0

#Telemetry data coming to this endpoint 
[UdpEndpoint listener]
Mode = Normal
Address = 224.1.2.34
Port = 14000

#I want to duplicate telemetry data and send to this endpoint.
[UdpEndpoint sender]
Mode = Normal
Address = 127.0.0.1
Port = 14550
AllowMsgId=0
```

jbeyerstedt, 2024-10-20:
If you want to use just UDP endpoints, one needs to be of the "server" mode so that the client can send data to it. The config you posted on the other hand will send to 127.0.0.1:14550, but not listen on that port.

emogooo, 2024-10-21:
As a result, what do you recommend I do?

jbeyerstedt, 2024-10-21:
You barely provided any information what connections you want to make, so I can only give generic answers.
As mentioned in my last post, your config needs to look like this:
```conf
[General]
Log=/mnt/ep
LogMode=always
TcpServerPort=0

[UdpEndpoint listener]
Mode = Normal
Address = 224.1.2.34
Port = 14000

[UdpEndpoint sender]
Mode = Server
Address = 127.0.0.1
Port = 14550
```
(By the way `AllowMsgId` is not a valid config item. You need to specify, if the filter should apply to the incoming or outgoing messages, see https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/examples/config.sample#L123 or a few lines further for incoming.)

jbeyerstedt, 2024-12-10:
Closing because of no activity. Feel free to comment and re-open, if there are things left to discuss.

closed 2024-12-10 by jbeyerstedt