mavlink-router/mavlink-router #436: Stack-based buffer overflow in ConfFile::get_sections

cktii, 2024-10-28:
Hi there,

I just happened to stumble upon an instance of a stack-based buffer overflow in the title-mentioned [function](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/common/conf_file.cpp#L435).

## Bug

```c
#define MAX_SECTION_NAME 100

// NOTE: This is a "public API"
int ConfFile::get_sections(const char *pattern, struct section_iter *iter)
{
    struct section *s;
    char section_name[MAX_SECTION_NAME]; // NOTE: Fixed sized initialization

    assert(pattern);
    assert(iter);

    if (iter->ptr == nullptr) {
        s = _sections;
    } else {
        s = ((section *)iter->ptr)->next;
    }

    for (; s; s = s->next) {
        memcpy(section_name, s->name, s->len);  // BUG: Copies s->len (user-controlled) into a fixed size buffer
        section_name[s->len] = 0; // BUG: Appends a NULL bytes that can also lead to an out-of-bounds write
        if (fnmatch(pattern, section_name, FNM_CASEFOLD) == 0) {
            iter->name = s->name;
            iter->name_len = s->len;
            iter->ptr = s;
            return 0;
        }
    }

    iter->ptr = nullptr;
    iter->name = nullptr;
    iter->name_len = 0;
    return -ENOENT;
}
```

### Possible fix

I am by no means in expert for this project, but something like this mitigates the problem:

```diff
int ConfFile::get_sections(const char *pattern, struct section_iter *iter)
{
    struct section *s;
    char section_name[MAX_SECTION_NAME];

    assert(pattern);
    assert(iter);

    if (iter->ptr == nullptr) {
        s = _sections;
    } else {
        s = ((section *)iter->ptr)->next;
    }

    for (; s; s = s->next) {
+       uint32_t max_copy = s->len ? s->len < MAX_SECTION_NAME - 1 : MAX_SECTION_NAME - 1;
-       memcpy(section_name, s->name, s->len);
-       section_name[s->len] = 0;
+       memcpy(section_name, s->name, max_copy);
+       section_name[max_copy] = 0;
        if (fnmatch(pattern, section_name, FNM_CASEFOLD) == 0) {
            iter->name = s->name;
-           iter->name_len = s->len;
+           iter->name_len = max_copy;
            iter->ptr = s;
            return 0;
        }
    }

    iter->ptr = nullptr;
    iter->name = nullptr;
    iter->name_len = 0;
    return -ENOENT;
}
```

Alternatively, something like this if an early abort is preferred:

```c
if (s->len >= MAX_SECTION_NAME) {
    return -ENAMETOOLONG;
}
```

## Impact

Given that this is a "public" API and [any stack based buffer overflow is bad](https://cwe.mitre.org/data/definitions/121.html), this should get a fix ASAP. Furthermore, the bug is easy to trigger without much of a setup, and my fuzzing setup was able to find it quickly. 

## Details

```
 ./conf_parser.fuzzer ./crash-97ad93a12154e9bccba59ce56869d66baa84cead -detect_leaks=0
INFO: Running with entropic power schedule (0xFF, 100).
INFO: Seed: 2060636063
INFO: Loaded 1 modules   (2591 inline 8-bit counters): 2591 [0x630d18d0df18, 0x630d18d0e937),
INFO: Loaded 1 PC tables (2591 PCs): 2591 [0x630d18d0e938,0x630d18d18b28),
./conf_parser.fuzzer: Running 1 inputs 1 time(s) each.
Running: ./crash-97ad93a12154e9bccba59ce56869d66baa84cead
src/common/conf_file.cpp:452:9: runtime error: index 100 out of bounds for type 'char[100]'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior src/common/conf_file.cpp:452:9
=================================================================
==970129==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x742003009684 at pc 0x630d18c90f8c bp 0x7ffe5fc96e10 sp 0x7ffe5fc96e08
WRITE of size 1 at 0x742003009684 thread T0
    #0 0x630d18c90f8b in ConfFile::get_sections(char const*, ConfFile::section_iter*) /home/user/mavlink-router/src/common/conf_file.cpp:452:30
    #1 0x630d18cae3cc in LLVMFuzzerTestOneInput /home/user/mavlink-router/conf_parser_harness.cc:206:17
    #2 0x630d18b91ed4 in fuzzer::Fuzzer::ExecuteCallback(unsigned char const*, unsigned long) (/home/user/mavlink-router/conf_parser.fuzzer+0x69ed4) (BuildId: c4a43c6345161dacf44d55063014ce93e27d36c7)
    #3 0x630d18b7b006 in fuzzer::RunOneTest(fuzzer::Fuzzer*, char const*, unsigned long) (/home/user/mavlink-router/conf_parser.fuzzer+0x53006) (BuildId: c4a43c6345161dacf44d55063014ce93e27d36c7)
    #4 0x630d18b80aba in fuzzer::FuzzerDriver(int*, char***, int (*)(unsigned char const*, unsigned long)) (/home/user/mavlink-router/conf_parser.fuzzer+0x58aba) (BuildId: c4a43c6345161dacf44d55063014ce93e27d36c7)
    #5 0x630d18bab276 in main (/home/user/mavlink-router/conf_parser.fuzzer+0x83276) (BuildId: c4a43c6345161dacf44d55063014ce93e27d36c7)
    #6 0x74200502a1c9 in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #7 0x74200502a28a in __libc_start_main csu/../csu/libc-start.c:360:3
    #8 0x630d18b75bd4 in _start (/home/user/mavlink-router/conf_parser.fuzzer+0x4dbd4) (BuildId: c4a43c6345161dacf44d55063014ce93e27d36c7)

Address 0x742003009684 is located in stack of thread T0 at offset 132 in frame
    #0 0x630d18c9025f in ConfFile::get_sections(char const*, ConfFile::section_iter*) /home/user/mavlink-router/src/common/conf_file.cpp:436

  This frame has 1 object(s):
    [32, 132) 'section_name' (line 438) <== Memory access at offset 132 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/user/mavlink-router/src/common/conf_file.cpp:452:30 in ConfFile::get_sections(char const*, ConfFile::section_iter*)
Shadow bytes around the buggy address:
  0x742003009400: f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5
  0x742003009480: f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5
  0x742003009500: f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5
  0x742003009580: f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5 f5
  0x742003009600: f1 f1 f1 f1 00 00 00 00 00 00 00 00 00 00 00 00
=>0x742003009680:[04]f3 f3 f3 f3 f3 f3 f3 00 00 00 00 00 00 00 00
  0x742003009700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x742003009780: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x742003009800: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x742003009880: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x742003009900: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==970129==ABORTING
```

0xricksanchez, 2024-11-25:
This is now tracked under CVE-2024-51439

cktii, 2024-11-25:
This is now tracked under CVE-2024-51439

jbeyerstedt, 2024-11-25:
This is not exploitable in the mavlink-router application though!

Yes, `memcpy(section_name, s->name, s->len);` is not checking if `s->len` actually fits in the `section_name` char-buffer, but the code which fills/ creates the section structure is doing that (see [src/common/conf_file.cpp#L212](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/common/conf_file.cpp#L212)). So this can't be exploited using the application's interfaces (e.g. a specially crafted config file).

So how is this an actual vulnerability of this application? The module with the "public API" isn't even compiled as a (shared/ static) library, so it could only be exploited by writing new code which manipulates the config sections structs after reading the config file.