mavlink-router/mavlink-router #437: Various instances of NULL-PTR dereferences

cktii, 2024-10-28:
Similarly to #436 I also noticed a few instances where a [NULL-PTR dereference](https://cwe.mitre.org/data/definitions/476.html) due to a malformed `buffer` struct is occurring. The entry point for all of them were `Ulog::write_msg()`, `BinLog::write_msg()`.

## Occurrences

1. `Ulog::write_msg()` - [Ref.](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/ulog.cpp#L105)

```c
int ULog::write_msg(const struct buffer *buffer)
{
    const bool mavlink2 = buffer->data[0] == MAVLINK_STX;  // BUG: No guarantee that buffer is != NULL
    [...]
    case MAVLINK_MSG_ID_LOGGING_DATA_ACKED: {
        auto *ulog_data_acked = (mavlink_logging_data_acked_t *)buffer->curr.payload; // BUG: No guarantee that buffer->curr.paylouad != NULL
    [...]

```

2. `Ulog::_loggin_data_process()` - [Ref.](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/ulog.cpp#L228)

```c
void ULog::_logging_data_process(mavlink_logging_data_t *msg)
{
    bool drops = false;

    if (!_logging_seq(msg->sequence, &drops)) { // BUG: No guarantee that msg || msg->sequence are != NULL
        return;
    }
    [...]
```

3. `BinLog::write_msg()` - [Ref.](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/binlog.cpp#L85)

```c
int BinLog::write_msg(const struct buffer *buffer)
{
    const bool mavlink2 = buffer->data[0] == MAVLINK_STX; // BUG: No guarantee that buffer is != NULL
    [...]
```
 
4. `BinLog::_logging_data_process()` - [Ref.](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/binlog.cpp#L198)

```c
void BinLog::_logging_data_process(mavlink_remote_log_data_block_t *msg)
{
    ssize_t r;

    if (lseek(_file, msg->seqno * MAVLINK_MSG_REMOTE_LOG_DATA_BLOCK_FIELD_DATA_LEN, SEEK_SET) < 0) { // BUG: No guarantee that msg->seqno != NULL
    
    [...]
```

5. `AutoLog::write_msg()` - [Ref.](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/autolog.cpp#L42)

```c
int AutoLog::write_msg(const struct buffer *buffer)
{
    [...]

    const mavlink_heartbeat_t *heartbeat = (mavlink_heartbeat_t *)buffer->curr.payload; // BUG: Blind type-casting with no check for heartbeat != NULL

    [...]
```

## Impact 

In our case, this crashes the whole process and causes a form of local DoS.

## Fix

As all of them stem from the same root-cause (no sanity checks when handling user-/untrusted-input, or type casting) a simple NULL check coupled with an early return seems to solve all of these.

## Completeness 

These aforementioned instances we actively triggered with a fuzzer from our end. There's no guarantee there are not more instances like that throughout the code base.

cktii, 2025-01-02:
This issue has been assigned CVE-2024-53317

jbeyerstedt, 2025-01-02:
If you want to improve code quality, please just open a PR with a possible fix. Otherwise I'm not sure what your mission is here.