mavlink-router/mavlink-router #438: Buffer out-of-bounds read due to blind type-casting in AutoLog::write_msg

cktii, 2024-10-31:
Hi there,

I just happened to stumble upon an instance of a potential [out-of-bounds read](https://cwe.mitre.org/data/definitions/125.html) in the title-mentioned [function](https://github.com/mavlink-router/mavlink-router/blob/c20337bbc61abda21463e7482172d58b654e66b6/src/autolog.cpp#L42).

## Bug
The bug manifests in the following code:

```c
#define MAX_SECTION_NAME 100

// NOTE: This is a "public API"
int AutoLog::write_msg(const struct buffer *buffer) {
    [...]
    // BUG: Blindly trusting that buffer->curr.payload is large enough to hold a mavlink_heartbeat_t sized strucr
    const mavlink_heartbeat_t *heartbeat = (mavlink_heartbeat_t *)buffer->curr.payload;
    // NOTE: Following that we just access that type casted memory, which could be out of bounds for very small payloads

    /* We check autopilot on heartbeat */
    log_debug("Got autopilot %u from heartbeat", heartbeat->autopilot);
    if (heartbeat->autopilot == MAV_AUTOPILOT_PX4) {
  
```

### Possible fix

I am by no means in expert on this project, but something like this mitigates the problem and adds some more stability guarantees to the code:

```diff
int AutoLog::write_msg(const struct buffer *buffer)
{
    if (_logger != nullptr) {
        return _logger->write_msg(buffer);
    }

    /* set the expected system id to the first autopilot that we get a heartbeat from */
    if (_target_system_id == -1 && buffer->curr.msg_id == MAVLINK_MSG_ID_HEARTBEAT
        && buffer->curr.src_compid == MAV_COMP_ID_AUTOPILOT1) {
        _target_system_id = buffer->curr.src_sysid;
    }

    if (buffer->curr.msg_id != MAVLINK_MSG_ID_HEARTBEAT
        || buffer->curr.src_sysid != _target_system_id) {
        return buffer->len;
    }

+    if (buffer->curr.payload_len < sizeof(mavlink_heartbeat_t)) {
+       return buffer->len;
+    }

    const mavlink_heartbeat_t *heartbeat = (mavlink_heartbeat_t *)buffer->curr.payload;

+   if (!heartbeat) {
+       return buffer->len;
+   }

    /* We check autopilot on heartbeat */
    log_debug("Got autopilot %u from heartbeat", heartbeat->autopilot);
    if (heartbeat->autopilot == MAV_AUTOPILOT_PX4) {
        _logger = std::unique_ptr<LogEndpoint>(new ULog(_config));
    } else if (heartbeat->autopilot == MAV_AUTOPILOT_ARDUPILOTMEGA) {
        _logger = std::unique_ptr<LogEndpoint>(new BinLog(_config));
    } else {
        log_warning("Unidentified autopilot, cannot start flight stack logging");
    }

    return buffer->len;
}
```

## Impact
[... 84 lines cut ...]
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1575168==ABORTING
MS: 2 ChangeBinInt-CMP- DE: "\000\003"-; base unit: 211766218a3d76bd041ae73595eeb67779b5f231
0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xcd,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0xf7,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0x4e,0xf7,0x47,0x47,0x47,0x47,0x47,0x47,0x47,0x47,0x47,0x47,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0x0,0xe6,0xe6,0xe6,0x1a,0x19,0xe6,0xe6,0xe6,0xe6,0xf7,0xf7,0xf7,0x1,0xff,0x0,0x30,0x0,0x0,0x0,0x0,0x0,0x0,0x60,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xe6,0xe6,0xe6,0xe6,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x8,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xee,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xea,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xf7,0xf7,0xff,0xe6,0xe6,0xe6,0xe6,0xe6,0x3,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0xe6,0x0,0x0,0x0,0x0,0x0,0x5d,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xf7,0xf7,0x0,0x3,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xf7,0x1,0xff,0xff,0x0,0x0,0x0,0x0,0x0,0x0,0xff,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,
\367\367\367\367\367\367\367\367\367\367\367\367\315\367\367\367\367\367\367\367\367\367\367\367\367\367\367\367NNNNNNNNNNN\367GGGGGGGGGG\346\346\346\346\346\346\000\346\346\346\032\031\346\346\346\346\367\367\367\001\377\0000\000\000\000\000\000\000`\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\346\346\346\346\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\010\000\000\000\000\000\000\000\000\000\356\000\000\000\000\000\000\000\000\352\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\346\346\346\346\346\346\346\346\346\346\346\346\346\346\367\367\377\346\346\346\346\346\003\346\346\346\346\346\346\346\346\346\000\000\000\000\000]\000\000\000\000\000\000\000\000\000\000\000\000\000\367\367\000\003\000\000\000\000\000\000\000\367\001\377\377\000\000\000\000\000\000\377\000\000\000\000\000\000\000\000\000\000\000\000
artifact_prefix='./'; Test unit written to ./crash-a93724e4417872492a72d28ce7dfe91886ab2329
Base64: 9/f39/f39/f39/f3zff39/f39/f39/f39/f3905OTk5OTk5OTk5O90dHR0dHR0dHR0fm5ubm5uYA5ubmGhnm5ubm9/f3Af8AMAAAAAAAAGAAAAAAAAAAAAAAAAAAAAAAAAAAAObm5uYAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAAAAAAAAADuAAAAAAAAAADqAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5ubm5ubm5ubm5ubm5ub39//m5ubm5gPm5ubm5ubm5uYAAAAAAF0AAAAAAAAAAAAAAAAA9/cAAwAAAAAAAAD3Af//AAAAAAAA/wAAAAAAAAAAAAAAAA==

```

cktii, 2024-11-01:
## Addition

The aforementioned problem does not just affect `AutoLog`. It also seems to manifest in `ULog` on multiple occurrences. For example:

```c
int ULog::write_msg(const struct buffer *buffer) {
    [...]
    case MAVLINK_MSG_ID_LOGGING_DATA_ACKED: {
        // BUG: Another instance of casting to a struct without checking that buffer->curr.payload is has the appropriate size
        auto *ulog_data_acked = (mavlink_logging_data_acked_t *)buffer->curr.payload;
        mavlink_message_t msg;
        mavlink_logging_ack_t ack;

        // BUG: Accessing the new struct can be OOB
        ack.sequence = ulog_data_acked->sequence;
        ack.sequence = ulog_data_acked->sequence;
        ack.target_component = MAV_COMP_ID_ALL;
        ack.target_system = _target_system_id;
        // BUG: Providing the casted struct to this function as well
        mavlink_msg_logging_ack_encode(LOG_ENDPOINT_SYSTEM_ID, MAV_COMP_ID_ALL, &msg, &ack);
        _send_msg(&msg, _target_system_id);
        /* message will be handled by MAVLINK_MSG_ID_LOGGING_DATA case */

        if (trimmed_zeros) {
            mavlink_logging_data_t ulog_data;
            memcpy(&ulog_data, buffer->curr.payload, payload_len);
            memset(((uint8_t *)&ulog_data) + payload_len, 0, trimmed_zeros);
            _logging_process(&ulog_data);
        } else {
            // BUG: Another instance of a blind cast without a payload size check
            auto *ulog_data = (mavlink_logging_data_t *)buffer->curr.payload;
            _logging_process(ulog_data);
        }
        break;
    }
    case MAVLINK_MSG_ID_LOGGING_DATA: {
        if (trimmed_zeros) {
            mavlink_logging_data_t ulog_data;
            memcpy(&ulog_data, buffer->curr.payload, payload_len);
            memset(((uint8_t *)&ulog_data) + payload_len, 0, trimmed_zeros);
            _logging_data_process(&ulog_data);
        } else {
            // BUG: Another instance of a blind cast without a payload size check
            auto *ulog_data = (mavlink_logging_data_t *)buffer->curr.payload;
            _logging_data_process(ulog_data);
        }
        break;
    [...]
}

```

The called function basically just relays the potentially invalid struct members to `mavlink_msg_logging_ack_pack`

```c
/**
 * @brief Encode a logging_ack struct
 *
 * @param system_id ID of this system
 * @param component_id ID of this component (e.g. 200 for IMU)
 * @param msg The MAVLink message to compress the data into
 * @param logging_ack C-struct to read the message contents from
 */
static inline uint16_t mavlink_msg_logging_ack_encode(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg, const mavlink_logging_ack_t* logging_ack)
{
    return mavlink_msg_logging_ack_pack(system_id, component_id, msg, logging_ack->target_system, logging_ack->target_component, logging_ack->sequence);
}

/**
 * @brief Pack a logging_ack message
[... 269 lines cut ...]
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1558999==ABORTING
MS: 4 PersAutoDict-CopyPart-PersAutoDict-PersAutoDict- DE: "\3115"-"\025'\010\000 P\000\000"-"\013\001"-; base unit: f226bbc9a514dbf8714f8dd6297c5f5dd91297ad
0x0,0xff,0xff,0xff,0xff,0xce,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xc9,0x35,0xc9,0xff,0xff,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xff,0x2,0x0,0x92,0x90,0x92,0x0,0xef,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x18,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x3b,0x0,0x0,0x15,0x27,0x8,0x0,0x20,0x50,0x0,0x0,0x0,0x0,0x0,0xff,0xff,0xff,0xff,0x0,0x3,0xff,0xff,0xff,0xff,0xce,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xc9,0x35,0xc9,0xff,0xff,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xff,0x2,0x0,0x92,0x90,0x92,0x0,0xef,0x0,0x0,0x0,0x0,0x0,0xb,0x1,0x0,0x0,0x18,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xfd,0xfe,0x0,0x0,0xff,0x0,0xce,0x38,
\000\377\377\377\377\316\377\377\377\377\377\377\377\377\377\377\3115\311\377\377\000\000\000\000\000\000\000\000\377\002\000\222\220\222\000\357\000\000\000\000\000\000\000\030\000\000\000\000\000\000\000\000;\000\000\025'\010\000 P\000\000\000\000\000\377\377\377\377\000\003\377\377\377\377\316\377\377\377\377\377\377\377\377\377\377\3115\311\377\377\000\000\000\000\000\000\000\000\377\002\000\222\220\222\000\357\000\000\000\000\000\013\001\000\000\030\000\000\000\000\000\000\000\000\000\375\376\000\000\377\000\3168
artifact_prefix='./'; Test unit written to ./crash-5ae9a7d4a9a9026684e57d793d0fbbf987815f4d
Base64: AP/////O/////////////8k1yf//AAAAAAAAAAD/AgCSkJIA7wAAAAAAAAAYAAAAAAAAAAA7AAAVJwgAIFAAAAAAAP////8AA//////O/////////////8k1yf//AAAAAAAAAAD/AgCSkJIA7wAAAAAACwEAABgAAAAAAAAAAAD9/gAA/wDOOA==
```

cktii, 2025-01-02:
This issue has been assigned CVE-2024-53318

jbeyerstedt, 2025-01-02:
If you want to improve code quality, please just open a PR with a possible fix. Otherwise I'm not sure what your mission is here.