mavlink-router/mavlink-router #442: Not possible to install on NanoPi-NEO3 1GB RAM

Alina389, 2024-12-09:
Hello everyone,

The installation process fail (hang with max. cpu usage) at "ninja -C build" command, please check:

Script started on 2024-12-08 16:23:39+02:00 [TERM="xterm" TTY="/dev/pts/0" COLUMNS="237" LINES="59"]
[?2004h]0;pi@NanoPi-NEO3: ~/mavlink-routerpi@NanoPi-NEO3:~/mavlink-router$ [7mexport VERBOSE=1[27mexport VERBOSE=1
[?2004l
[?2004h]0;pi@NanoPi-NEO3: ~/mavlink-routerpi@NanoPi-NEO3:~/mavlink-router$ [7mninja -C build[27mninja -C build
[?2004l
ninja: Entering directory `build'

[0/25] Generating git_version.h with a custom command[K
[0/25] Compiling C++ object src/common/libcommon-private.a.p/conf_file.cpp.o[K
[0/25] Compiling C++ object src/common/libcommon-private.a.p/log.cpp.o[K
[0/25] Compiling C++ object src/common/libcommon-private.a.p/util.cpp.o[K
[0/25] Compiling C++ object src/common/libcommon-private.a.p/xtermios.cpp.o[K
[0/25] Compiling C++ object examples/arm-authorizer.p/arm-authorizer.cpp.o[K
[1/25] Compiling C++ object src/common/libcommon-private.a.p/xtermios.cpp.o[K
[1/25] Compiling C++ object examples/heartbeat-print.p/heartbeat-print.cpp.o[K
[2/25] Generating git_version.h with a custom command[K
[2/25] Compiling C++ object src/mavlink-routerd.p/autolog.cpp.o[K
[3/25] Compiling C++ object src/common/libcommon-private.a.p/log.cpp.o[K
[3/25] Compiling C++ object src/mavlink-routerd.p/binlog.cpp.o[K
[4/25] Compiling C++ object src/common/libcommon-private.a.p/util.cpp.o[K
[4/25] Compiling C++ object src/mavlink-routerd.p/dedup.cpp.o[K
[5/25] Compiling C++ object src/common/libcommon-private.a.p/conf_file.cpp.o[K
[5/25] Linking static target src/common/libcommon-private.a[K
[6/25] Linking static target src/common/libcommon-private.a[K
[6/25] Compiling C++ object src/mavlink-routerd.p/endpoint.cpp.o[K
[7/25] Compiling C++ object src/mavlink-routerd.p/dedup.cpp.o[K
[7/25] Compiling C++ object src/mavlink-routerd.p/git_version.cpp.o[K
[8/25] Compiling C++ object examples/heartbeat-print.p/heartbeat-print.cpp.o[K
[8/25] Compiling C++ object src/mavlink-routerd.p/logendpoint.cpp.o[K
[9/25] Compiling C++ object src/mavlink-routerd.p/git_version.cpp.o[K
[9/25] Compiling C++ object src/mavlink-routerd.p/main.cpp.o[K
[10/25] Compiling C++ object examples/arm-authorizer.p/arm-authorizer.cpp.o[K
[10/25] Compiling C++ object src/mavlink-routerd.p/mainloop.cpp.o[K
[11/25] Compiling C++ object src/mavlink-routerd.p/autolog.cpp.o[K
[11/25] Compiling C++ object src/mavlink-routerd.p/pollable.cpp.o[K
[12/25] Compiling C++ object src/mavlink-routerd.p/pollable.cpp.o[K
[12/25] Compiling C++ object src/mavlink-routerd.p/timeout.cpp.o[K
[13/25] Compiling C++ object src/mavlink-routerd.p/binlog.cpp.o[K
[13/25] Compiling C++ object src/mavlink-routerd.p/ulog.cpp.o[K
[14/25] Compiling C++ object src/mavlink-routerd.p/timeout.cpp.o[K
[14/25] Compiling C++ object src/mavlink-routerd.p/tlog.cpp.o[K^C^C
ninja: build stopped: interrupted by user.

[?2004h]0;pi@NanoPi-NEO3: ~/mavlink-routerpi@NanoPi-NEO3:~/mavlink-router$ exit
[?2004l
exit

Script done on 2024-12-08 16:25:17+02:00 [COMMAND_EXIT_CODE="130"]

I already tried to install the v3 version, there is the nearly same issue, installation process fail (hang with max. cpu usage) at "ninja -C build" command:
pi@NanoPi-NEO3:~/mavlink-router$ ninja -C build
ninja: Entering directory `build'
[16/24] Compiling C++ object examples/px4-offboard-mode.p/px4-offboard-mode.cpp.o

Please give me some advice, it's looks like memory overflow as I guess.

Thank you.

jbeyerstedt, 2024-12-09:
As already discussed in multiple other issues and discussion threads (e.g. #418, #380, #381, #319, #349, #320), this is probably related to insufficient RAM.
I would highly recommend to cross-compile. Otherwise you can try to add some swap or check if compiling single-threaded (add `-j 1`) helps.

Alina389, 2024-12-09:
> As already discussed in multiple other issues and discussion threads (e.g. #418, #380, #381, #319, #349, #320), this is probably related to insufficient RAM. I would highly recommend to cross-compile. Otherwise you can try to add some swap or check if compiling single-threaded (add `-j 1`) helps.

Thank you for the fast, correct help.
The "sudo ninja -j1 -C build" command helped, successfully compiled.

closed 2025-01-07 by jbeyerstedt