mavlink-router/mavlink-router #444: Mavlink-router SERVER configuration in details

nforest92, 2024-12-12:
Hello Guys,

I'm lost in mavlink-router configuration, please point me to the right way.
I already read the documentation here, but a few things unclear.
My configuration is: UAV flight controller -(UART)-> mavlink-router client -(LTE modem)-> mavlink-router server -(LTE modem)-> mission planner
As you see only the "mavlink-router server" have static IPv4 address, both "mavlink-router client" and mission-planner have dynamic IP addresses, also they have some network latency about 200-300 msec as default, however the UAV can flight where poorly covered LTE then the latency can be a few seconds also.
My "mavlink-router client" configuration looks like:
 [UdpEndpoint local]
   Mode=normal
   Address=ip_address_of_the_server
   Port=14550

[UartEndpoint serial1]
   Device=/dev/ttyS1
   Baud=57600
   
How I need to configure the "mavlink-router server"?
Please give me some example based on my provided request.
I searched a lot on internet, I don't fund any detailed information about the mavlink-router server configuration, only a few copy of the official description which can be good for experienced users but not for new bees as I'm.
I'm sure your detailed explanation will help for much others who will find here my question.

Thanks a lot.

terence-berak, 2024-12-13:
> Hello Guys,
>
> I'm lost in mavlink-router configuration, please point me to the right way. I already read the documentation here, but a few things unclear. My configuration is: UAV flight controller [...]

I would be interested in a similar situation!
Thank you,
Terence

dakejahl, 2024-12-13:
AFAIK it's not really feasible to directly connect two endpoints via IP address over LTE due to IP assignment from the ISP behind carrier grade NAT. The solution I found is to use a VPN server which both endpoints connect to. This allows your VPN to provide the VLAN for the devices and you can assign static IPs to each connection.

lucasdemarchi, 2024-12-13:
You will need one of those with a public IP so you can reach it from the other side. If you don't, then you can setup VPS or the like that allows to connect the 2.

As mentioned above a VPN would work too. I've heard from people at https://github.com/uavpal/disco4g  they had good experiences with zerotier for a simpler setup. There are also other similar services available from other companies like tailscale.

nforest92, 2024-12-13:
> You will need one of those with a public IP so you can reach it from the other side. If you don't, then you can setup VPS or the like that allows to connect the 2.

Yes of course, the "mavlink-router server" installed on a linux based VPS with static IPv4 address.
I looking for the "main.conf" data, how to configure the server to accept heartbeat (connection) from the UAV side mavlink-router client let's say on the port 14550 of the VPS's IPv4 and "proxy" the telemetry stream to another port, where I can connect with the mission-planner software use the UDP ports.
As I read (and good understand) the README file the mavlink-router program capable to do this types of connections.
I also need a bit help, how to fine tune the latency issue what I described above.

Connect everything on the same VPN isn't the best option for me, while need take extra care about the VPN connections issues, wait for internet access then wait for VPN connection established there after wait for mavlink-router connection. Also when the UAV lost the GSM/LTE signal then take care about the re-connections.

Mavlink-router brilliantly designed to take care about the connection issues as I read in the README file, however I can't configure alone.

Please provide me an example "main.conf" data for the server side.

Thanks!