mavlink-router/mavlink-router #452: Please help me understand how to configure my routing

zdanek, 2025-02-15:
Hi.
I'm struggling a bit with proper config, probably because I'm still not familiar with MR and also, that a part of the system was written by me, so it might not obey general mavlink rules.

I have 3 sets of drones. Each drone has: pixhawk with Ardupilot, camera on gimbal that has proprietary steering command set, good radio link that connects to shared mesh network. 
At ground I have a server that for each camera, has own instance of software (written by me), to translate RC channels to the steering commands.
There is also QGC as Ground station. Which, by the way, is using 255,190.

So the work is easy - my software listens for RC channels and sends camera commands accordingly.

A few months ago, all camera software was set on Raspberry Pi, as onboard computer, attached via serial link. It worked like a charm. Just got a copy of all mavlink in-out of Pixhawk.
Now I had to move it to ground and use mesh network to get mavlink and send camera commands.

Now hardest (for me) part:
1) drones have system ids 1, 2, 3 and component id 1 for each - (1 is for onboard FC.)
2) I set for each camera software system Id 1, 2 and 3 accordingly and 191 as component Id (191 is for onboard computer).
3) each drone has own UDP endpoint, as well as my camera software instances  - 3 udp endpoints.

Drones work well with QGC. No lag, bi-directional comm.
Camera software suffers from lags. First I discovered that I need to send heartbeat in order to open communication. Now it's obvious, but bear in mind that previously it was just serial comm "that works". 

Is it possible that MavlinkRouter have problems with my "camera endpoints" because I use system id 1, 2, 3 - same as for drones, on other endpoints?
Shall I use groups? Will it send then all the traffic? Shall I change to other system id numbers, like 4,5,6 and expect that MR will send RC channels that are meant for 1,2,3? How?

Please help me, because I almost lost all my hair on that :-/
Should you have any other clarifying questions, please ask! Help! 🐝

zdanek, 2025-02-15:
@jbeyerstedt and/or @lucasdemarchi  maybe you could help me here? I'd appreciate it.

jbeyerstedt, 2025-02-16:
Do I understand it correctly that communication flow is like this:
```
gimbal <-(UART?)-> pixhawk <-(IP Mesh)-> PC/RPi [mavlink-router and camera SW]
```

So I assume each Pixhawk has its own static IP and your UDP endpoints to the drones are "normal" mode?
If so, you don't need endpoint groups as those are only needed for parallel/ redundant data links (which is not the case, if each drone has its own IP address).

And your software is sending MAVLink packets addressed to the Pixhawk or another component ID on the drone's system ID? Because it's important that your camera software is not using the same component ID as something on the drone. But the target system (gimbal on the drone in this case) also needs to send heartbeats. Otherwise mavlink-router does not know where to find the gimbal.

The routing rules are described in the [corresponding Readme section](https://github.com/mavlink-router/mavlink-router/blob/master/README.md#message-routing).
You can check which system and component IDs mavlink-router has detected on each endpoint by running it in verbose mode.
Because my first guess would be the loop prevention logic, even though in that case it shouldn't lag but just don't send at all.

zdanek, 2025-02-16:
@jbeyerstedt  thanks for trying to help.
Nope, it's a bit different:

[image]

Drone flies with camera. Both has own IP. 
Gimballer is my software to listen to RC channels and send camera commands.
Gimballer, MR and QGC are on the ground. Having own IP. There is no UART. Only UDP.

I have 3 such drones. Thus the triad: Drone, Camera, Gimballer has 3 exact copies. 3 drones with own cameras and own instances of Gimballer to control each camera.

Each drone has distinct system Id 1, 2 ,3. 

The questions are for what system Ids choose for gimballers, to they can always get RC channels sent to each / by each drones.

jbeyerstedt, 2025-02-18:
Could you please answer my question about which component ID the gimbaler uses and if the camera subsystem sends heartbeats?
And a dump of you mavlink-router config would also help.

zdanek, 2025-02-18:
@jbeyerstedt 
Gimballer uses **same system_id** as it's drone so 1 or 2 or 3, depending on instance.
Compontent ID is companion computer, that is: MAV_COMP_ID_ONBOARD_COMPUTER=191

Camera is not visible to pixhawk and mavlink. It's separate device. It has own IP in the same network, but doesn't use mavlink. 
My Gimballer sends HB every second.

My concerns are shall Gimballer use same system id as drone. Gimballer is on separate endpoint.

Config:
```
[UdpEndpoint Drone_1]
Mode = Server
Address = 0.0.0.0
Port = 14551

[UdpEndpoint Drone_2]
Mode = Server
Address = 0.0.0.0
Port = 14552

[UdpEndpoint Drone_3]
Mode = Server
Address = 0.0.0.0
Port = 14553

[UdpEndpoint GCS]
Mode = Server
Address = 0.0.0.0
Port = 14550

[UdpEndpoint Gimballer_1]
Mode = Server
Address = 0.0.0.0
Port = 16551

[UdpEndpoint Gimballer_2]
Mode = Server
Address = 0.0.0.0
Port = 16552

[UdpEndpoint Gimballer_3]
Mode = Server
Address = 0.0.0.0
Port = 16553
```

zdanek, 2025-02-18:
@jbeyerstedt I've found the cause. I'm sorry for confusion, but I think this discussion might be valuable for other users. 

I have to admit, that the fault was mine. Gimballer is simply ineffective while reading and processing mavlink messages.

While it was on UART and got a narrow stream of messages it was performing good. Now, when ALL the messages from ALL drones were delivered to it, it  drowned in amount of processing. 

I got out of this thanks to wonderful MR config options - I limited routing for Gimballers to it's dedicated drones and limited to only 4 useful mavlink message types. Rest is filtered out by MR.

I will rewrite port support to make it more robust.

@jbeyerstedt Thank you for you willing to help. Sorry for taking your time :)

jbeyerstedt, 2025-02-18:
I would only be concerned, if Gimballer and the camera subsystem on the drone use the same component ID. Because that would be blocked by the loop prevention of mavlink-router. That's why I asked which component IDs are used.

But nice that the extensive filtering options helped in this case.

closed 2025-03-05 by jbeyerstedt