mavlink-router/mavlink-router #454: FAILED: meson-internal__install

crazy-robotics, 2025-03-12:
I am using `ubuntu 20.04` while doing `sudo ninja -C build install` i got following error
```
crazy@robotics:~/mavlink-router$ sudo ninja -C build install
ninja: Entering directory `build'
[1/2] Installing files
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 129, in run
    return options.run_func(options)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 514, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 330, in do_install
    d = pickle.load(ifile)
ModuleNotFoundError: No module named 'mesonbuild.options'
FAILED: meson-internal__install 
/usr/bin/meson install --no-rebuild
```

jbeyerstedt, 2025-03-12:
Did you install meson from the ubuntu packages? Because the version shipped in 20.04 is too old. See https://github.com/mavlink-router/mavlink-router?tab=readme-ov-file#compilation-and-installation:

> Note that meson package must be version 0.55 or later. If your package manager does not have this version, a more recent version can be downloaded via pip: sudo pip3 install meson

jbeyerstedt, 2025-05-15:
closing because of inactivity. Please just respond here if you need additional assistance.

closed 2025-05-15 by jbeyerstedt