mavlink-router/mavlink-router #457: Drop packages out sequence

pa-hagen, 2025-05-06:
Hi!

I have a redundant link system with dual LTE downlink from drone. I use mavlink-router on top side to distribute to two companion computers, each with separate 4G router. I use another mavlink-router instance on ground to gather all messages before deDuplication and forward to GCS. I utilize the group function, and it all works very well when i have good connection on both systems, or if one of them is fully down. 

When i have poor reception on one system, some of the packages can get very delayed - up to 10 seconds some times. This makes the UAV "jump" around on the screen, and delta values (like vertical speed) calculated in Mission Planner goes craczy. It also makes me concered in the (rare) case that we are using joystick/USB remote to control the UAV. 

I am open for other suggestion on how to mitigate this, but could it be possible to implement a way to drop messages if it has already received a message of same ID/SYS/COMP with an equal or larger sequence number? 

Best regards,
Pål Are

pa-hagen, 2025-05-08:
Thanks for the reply Jannik. 

I am using the deduplication at 800ms to filter out duplicates, but according to readme it is only recommended with <1 sec to not risc loosing packages. However, it also says the entire package, inluding sequence number has to be identical for it to be dropped, so i don't know how packages could be dropped unintentionally.

Would i have to be worried about high workload or overflow of some conainer if the deduplication period is set high? 10 seconds or more could be a lot of messages to compare for deduplication. 

PA:

jbeyerstedt, 2025-05-09:
> but according to readme it is only recommended with <1 sec to not risc loosing packages. However, it also says the entire package, inluding sequence number has to be identical for it to be dropped, so i don't know how packages could be dropped unintentionally.

Yeah, I don't remember exactly how I came up with that number. As stated in my previous message the only way the de-duplication can lead to falsely dropped messages is if the whole message content is the same (including sequence number), which should be quite unlikely. At least for normal telemetry as there will be quite some noise on the reported measurements. Only commands may have a higher risk as they don't contain measurements.

> Would i have to be worried about high workload or overflow of some container if the deduplication period is set high? 10 seconds or more could be a lot of messages to compare for deduplication.

The message hashes are stored in a hash-map without a maximum size. So it will "just" need more RAM, but even the lookup should be quite fast because it's a map and not a plain array/ vector.